libcsp/libcsp #536: switch on -Wall -Wextra flags

dimitrovand, 2024-03-02:
Do you think it's a good idea to start using '-Wall -Wextra' flags instead of specific options? Enabling these will help improve the code quality by preventing common mistakes. Going a step further, and treating all warnings as errors, will force everyone to handle warnings at the time they occur. 
Note that this update will require some changes in the source, I think all of them are related to explicitly handling unused variables, but it's worth spending a day or two on making the necessary changes.

yashi, 2024-03-02:
What do you mean? All three build systems use at least `-Wall -Wextra`.  Meson even enables `-Werror`.

dimitrovand, 2024-03-02:
You're correct, sorry. Got confused by the CSP_C_ARGS parameters in the main CMakeLists.txt. I'm closing it.

closed 2024-03-02 by dimitrovand

yashi, 2024-03-02:
No problem at all! Thank you for your time and efforts. :smiley: