libcsp/libcsp #544: build: differences in cmake, meson and waf

dimitrovand, 2024-03-20:
It seems like there are some differences in building `csp_rtable_cidr.c` and `csp_rtable_stdio.c`. 
In CMake and Waf everything looks identical, but Meson's build is slightly different:

In `src/meson.build`:
```
if get_option('have_stdio') and get_option('use_rtable')
	csp_sources += files('csp_rtable_stdio.c')
endif

if get_option('use_rtable')
	csp_sources += files('csp_rtable_cidr.c')
endif
```

In `src/CMakeLists.txt`:
```
target_sources(csp PRIVATE
  ...
  csp_rtable_cidr.c
  ...
)
...
if (CSP_HAVE_STDIO)
  target_sources(csp PRIVATE csp_rtable_stdio.c)
endif()
```

In `wscript`:
```
    ctx.env.append_unique('FILES_CSP', [...,
                                        'src/csp_rtable_cidr.c'])

    # Add if stdio
    if ctx.check(header_name="stdio.h", mandatory=False):
        ctx.define('CSP_HAVE_STDIO', True)
        ctx.env.append_unique('FILES_CSP', ['src/csp_rtable_stdio.c'])
```

yashi, 2024-03-20:
I think we can build `csp_rtable_cidr.c` unconditionally because it doesn't depdend on anything externally.  OTOH, to build `csp_rtable_stdio.c` the target must have `stdio.h`.

Compilers do a good job of removing unused fuctions and static data out from the final binary, anyway.

```diff
diff --git a/src/meson.build b/src/meson.build
index b9bf4df6..83e14b40 100644
--- a/src/meson.build
+++ b/src/meson.build
@@ -11,6 +11,7 @@ csp_sources += files([
        'csp_port.c',
        'csp_qfifo.c',
        'csp_route.c',
+       'csp_rtable_cidr.c',
        'csp_service_handler.c',
        'csp_services.c',
        'csp_id.c',
@@ -25,10 +26,6 @@ if get_option('have_stdio') and get_option('use_rtable')
        csp_sources += files('csp_rtable_stdio.c')
 endif
 
-if get_option('use_rtable')
-       csp_sources += files('csp_rtable_cidr.c')
-endif
-
 if get_option('enable_csp_print')
        csp_sources += files('csp_hex_dump.c')
 endif
```

### For native (dynamic linking):
```
❯ rm -rf builddir && meson setup builddir && ninja -C builddir examples/csp_client 
    :
❯ nm builddir/libcsp.so g rtable
0000000000008ac5 T csp_rtable_clear
00000000000089a1 T csp_rtable_find_route
0000000000008ab1 T csp_rtable_free
0000000000008b28 T csp_rtable_iterate
0000000000008b5f T csp_rtable_print
000000000000893a t csp_rtable_print_route
0000000000008978 T csp_rtable_search_backward
0000000000008aca T csp_rtable_set
0000000000008a32 T csp_rtable_set_internal
0000000000016aa0 b rtable
0000000000016a80 b rtable_inptr
❯ nm builddir/examples/csp_client g rtable
❯ 
```

### For Zephyr (static linking):
```
❯ nm build/modules/libcsp/build/libcsp.a g rtable
csp_rtable_cidr.c.obj:
00000000 T csp_rtable_clear
00000000 T csp_rtable_find_route
00000000 T csp_rtable_free
00000000 T csp_rtable_iterate
00000000 T csp_rtable_print
00000000 t csp_rtable_print_route
00000000 T csp_rtable_search_backward
00000000 T csp_rtable_set
00000000 T csp_rtable_set_internal
00000000 b rtable
00000000 b rtable_inptr
csp_rtable_stdio.c.obj:
00000000 T csp_rtable_check
         U csp_rtable_iterate
00000000 T csp_rtable_load
00000000 t csp_rtable_parse
00000000 T csp_rtable_save
00000000 t csp_rtable_save_route
         U csp_rtable_set
❯ nm build/zephyr/zephyr.elf g rtable
❯ 
```

Let me know if your case is different.

yashi, 2024-03-20:
or, to align with #517, we should make it conditional.

PR #555 (2024-04-20): Build csp rtable cidr conditionally
This fixes #544.
closed 2024-04-20 by yashi