libcsp/libcsp #547: Example on how to use I2C

curat97, 2024-03-27:
Hello, I am opening this issue to ask if there is any example on how to use the i2c interface. I am using the libcsp library for a project involving two raspberry pi zero. I was able to correctly send and receive data using the UDP and CAN interface, however I am having troubles configuring the i2c interface. 
I am using the function _csp_i2c_add_interface_ to add the i2c interface and the function _csp_sendto_ to send the data. When I execute my C program, I get a segmentation fault error in the function _csp_i2c_tx_. I attach here the code of the C program. I premise that I am not an expert in C programming, so I could make trivial mistakes, but I hope this issue could be useful to others trying to use the i2c interface. 

`#include <unistd.h>
#include <stdio.h>
#include <string.h>
#include <csp/csp.h>
#include <csp/csp_debug.h>
#include <stdlib.h>
#include <pthread.h>
#include <csp/interfaces/csp_if_i2c.h>
#include <csp/csp_crc32.h>

static unsigned int rpi_sent = 0;
uint32_t opts = CSP_O_CRC32;
uint8_t prio = CSP_PRIO_NORM;
uint16_t rpi_csp_Addr = 2000;
uint8_t rpi_csp_port = 12;	
uint16_t cubesat_I2C_csp_Addr = 4005;
uint8_t cubesat_I2C_csp_port = 20;

static int csp_pthread_create(void * (*routine)(void *)) {

	pthread_attr_t attributes;
	pthread_t handle;
	int ret;

	if (pthread_attr_init(&attributes) != 0) {
		return CSP_ERR_NOMEM;
	}
	/* no need to join with thread to free its resources */ 
	pthread_attr_setdetachstate(&attributes, PTHREAD_CREATE_DETACHED);

	ret = pthread_create(&handle, &attributes, routine, NULL);
	pthread_attr_destroy(&attributes);

	if (ret != 0) {
		return ret;
	}

	return CSP_ERR_NONE;
}

// Router task

static void * task_router(void * param) {
	while (1) {
		csp_route_work();
	}
	return NULL;
}

int router_start(void) {
	return csp_pthread_create(task_router);
}

// client task 

void client(void) {
    int j;
	for (j=0; j<5; j=j+1){

		csp_packet_t * packet = csp_buffer_get(100);		
		char exStr[] = "Hello world \r\n";
		memcpy(packet->data, exStr ,strlen(exStr)+1);
		packet->length = (strlen((char *) packet->data) +1);
		csp_crc32_append(packet);
		csp_sendto(prio,cubesat_I2C_csp_Addr,cubesat_I2C_csp_port,rpi_csp_port,opts,packet);
		if (csp_dbg_errno != 0) {
			csp_print("csp_sendto() failed with csp_dbg_errno=%u\n", csp_dbg_errno);
		}
		else {
			csp_print("Packet sent \n");
		}
		++rpi_sent;
		sleep(1);
	}
}

static void * task_client(void * param) {
	client();
	return NULL;
}

int client_start(void) {
	return csp_pthread_create(task_client);
}

// Add i2c interface

static csp_iface_t csp_ifadd_i2c(void)	{
    csp_iface_t *iface;
	iface =	malloc(sizeof(csp_iface_t));
    csp_i2c_interface_data_t *ifdata;
	ifdata = malloc(sizeof(csp_i2c_interface_data_t));
	ifdata->tx_func = csp_i2c_tx;
    iface->name = CSP_IF_I2C_DEFAULT_NAME;
    iface->addr = 4000;
    iface->interface_data = ifdata;
	
    int error = csp_i2c_add_interface(iface);
    if (error == CSP_ERR_INVAL) {
		csp_print("CSP_ERR_INVAL \n");
    }
    iface->is_default = 1;
    return *iface;
}

// main

int main() {
	
	/* Init CSP */
    csp_init();
    /* Start router */
	int error;
	
    error = router_start();
	if (error != CSP_ERR_NONE) {
		csp_print("error starting router \n");
		return 1 ;
	}
	else {
		csp_print("Router started\n");
	}
	
	csp_print("Adding I2C interface \n");
	csp_ifadd_i2c();
	csp_print("Interface list\r\n");
    csp_iflist_print();
	
	/* Start client thread */
	
    error = client_start();
	if (error != CSP_ERR_NONE) {
		csp_print("error starting client \n");
		return 1;
	}
	
    while(1) {
        sleep(3);
		csp_print("errorno: %d\n", csp_dbg_errno);
		csp_print("Rpi sent %u packets\n",rpi_sent);
		csp_iflist_print();
    }

	return 0;
}
`

yashi, 2024-03-28:
I've never used it.  #316 seems to port the interface driver to 2.0 but not sure.

> I get a segmentation fault error in the function csp_i2c_tx

It seems you have to give your own send function to `iface` when you call `csp_i2c_add_interface()` just like `csp_usart_open_and_add_kiss_interface()` does.

curat97, 2024-03-28:
Thank you for your suggestion. I managed to make it work in the following way. It may not be the best or the most elegant way but it does the job. Worth mentioning, I had to configure the other rpi as a i2c slave following the _LinusCDE_ instructions here (https://raspberrypi.stackexchange.com/questions/76109/raspberry-as-an-i2c-slave)

`#include <unistd.h>
#include <stdio.h>
#include <string.h>
#include <csp/csp.h>
#include <csp/csp_debug.h>
#include <stdlib.h>
#include <pthread.h>
#include <csp/interfaces/csp_if_i2c.h>
#include <csp/csp_crc32.h>
#include <pigpio.h>

static unsigned int rpi_sent = 0;
uint32_t opts = CSP_O_CRC32;
uint8_t prio = CSP_PRIO_NORM;
uint16_t rpi_csp_Addr = 2000;
uint8_t rpi_csp_port = 12;	
uint16_t cubesat_I2C_csp_Addr = 8;
uint8_t cubesat_I2C_csp_port = 16;

uint8_t i2cBus = 1;
uint8_t i2cFlags = 0;
int i2chandle;

static int csp_pthread_create(void * (*routine)(void *)) {

	pthread_attr_t attributes;
	pthread_t handle;
	int ret;

	if (pthread_attr_init(&attributes) != 0) {
		return CSP_ERR_NOMEM;
	}
	/* no need to join with thread to free its resources */ 
	pthread_attr_setdetachstate(&attributes, PTHREAD_CREATE_DETACHED);

	ret = pthread_create(&handle, &attributes, routine, NULL);
	pthread_attr_destroy(&attributes);

	if (ret != 0) {
		return ret;
	}

	return CSP_ERR_NONE;
}

// Router task

static void * task_router(void * param) {
	while (1) {
		csp_route_work();
	}
	return NULL;
}

int router_start(void) {
	return csp_pthread_create(task_router);
}

// client task 

void client(void) {
    int j;
	for (j=0; j<5; j=j+1){

		csp_packet_t * packet = csp_buffer_get(100);		
		char exStr[] = "Hello world \r\n";
		memcpy(packet->data, exStr ,strlen(exStr)+1);
		packet->length = (strlen((char *) packet->data) +1);
		csp_crc32_append(packet);
		csp_sendto(prio,cubesat_I2C_csp_Addr,cubesat_I2C_csp_port,rpi_csp_port,opts,packet);
		if (csp_dbg_errno != 0) {
			csp_print("csp_sendto() failed with csp_dbg_errno=%u\n", csp_dbg_errno);
		}
		else {
			csp_print("Packet sent \n");
		}
		++rpi_sent;
		sleep(1);
	}
}

static void * task_client(void * param) {
	client();
	return NULL;
}

int client_start(void) {
	return csp_pthread_create(task_client);
}

// Add i2c interface

int i2cWriteBlockDataWrapper(void * driver_data, csp_packet_t * frame) {
    return i2cWriteBlockData(i2chandle, cubesat_I2C_csp_Addr, (char *)frame->data, frame->length);
}

static csp_iface_t csp_ifadd_i2c(void)	{
    gpioTerminate();
    if (gpioInitialise() < 0) perror("error GPIO init");
    i2chandle = i2cOpen(i2cBus,cubesat_I2C_csp_Addr,i2cFlags);
    	
	csp_iface_t *iface;
	iface =	malloc(sizeof(csp_iface_t));
    csp_i2c_interface_data_t *ifdata;
	ifdata = malloc(sizeof(csp_i2c_interface_data_t));
	ifdata->tx_func = i2cWriteBlockDataWrapper;
    iface->name = CSP_IF_I2C_DEFAULT_NAME;
    iface->addr = 4000;
    iface->interface_data = ifdata;
	
    int error = csp_i2c_add_interface(iface);
    if (error == CSP_ERR_INVAL) {
		csp_print("CSP_ERR_INVAL \n");
    }
    iface->is_default = 1;
    return *iface;
}

// main

int main() {
	
	/* Init CSP */
    csp_init();
    /* Start router */
	int error;
	
    error = router_start();
	if (error != CSP_ERR_NONE) {
		csp_print("error starting router \n");
		return 1 ;
	}
	else {
		csp_print("Router started\n");
	}
	
	csp_print("Adding I2C interface \n");
	csp_ifadd_i2c();
	csp_print("Interface list\r\n");
    csp_iflist_print();
	
	/* Start client thread */
	
    error = client_start();
	if (error != CSP_ERR_NONE) {
		csp_print("error starting client \n");
		return 1;
	}
	
    while(1) {
        sleep(3);
		csp_print("errorno: %d\n", csp_dbg_errno);
		csp_print("Rpi sent %u packets\n",rpi_sent);
		csp_iflist_print();
    }

	return 0;
}
`

yashi, 2024-03-29:
Hi @curat97,

Thank you for confirming that I2C interface works with the current libcsp!  I'm closing this for now.

If you have any other problem, don't hesitate to open a new issue! Thanks you!

curat97, 2024-04-19:
Hi @yashi, sorry for reopening this so late but I wanted to point out that I only partially tested the I2C interface with libcsp. The code I provided to you will for sure send a CSP packet through the I2C interface from the master device to the slave device (both Raspberry Pi Zero W). On the slave side, I am not able to receive the packet by binding a callback to the I2C port (while this works with other interface ports, e.g. CAN, UDP) but I can read the packet using the pigpio bscXfer function. On the other side, I was not able to transmit a CSP packet from slave to master, this seems to be more like a [raspberry issue ](https://raspberrypi.stackexchange.com/questions/108780/raspberry-pi-as-i2c-slave-problem-sending-data-back). I guess that with some coding one can make it to work. 
All this to say that I am not sure if it's safe to say that the I2C interface has been tested with CSP. Sorry again for pointing this out so late.

yashi, 2024-04-19:
@curat97 Thanks for the info! I'll reopening this.

yashi, 2025-10-16:
Just like USB, you (as the controller) periodically poll the target devices and initiate reads from them.