libcsp/libcsp #551: CSP_INIT returns CSP_ERROR_NOMEM

hernaangq, 2024-04-17:
Hello, 
I am using CSP in a EFM32 board and I have it correctly linked in my project. 
But, when calling csp_init (&csp_conf) it returns CSP_ERROR_NOMEM. 

HOW can I fix that??
Thanks.

Hernan

hernaangq, 2024-04-17:
``
void CSP_Init(){
    Uint8_t address = 2;
    
    csp_conf_t csp_conf;
    csp_conf_get_defaults(&csp_conf);
    csp_conf.address = address;

    int error = csp_init(&csp_conf);
    if (error != CSP_ERR_NONE){
     //   debug_printf_safe(ERROR,"CSP INIT ERROR %d",error);
    }

    csp_route_start_task(500, 0);
}
``

moonlight83340, 2024-04-18:
what version of csp are you using ?
Can you send your code here ?
The `CSP_ERR_NOMEM` means a memory error.

yashi, 2024-04-18:
@hernaangq, it appears you are using v1.6 or a pre-v2.0 version since `csp_init()` is returning int.

In v1.6, many initialization functions use `csp_malloc()` and `csp_calloc()` to allocate memory.

- https://github.com/libcsp/libcsp/blob/libcsp-1/src/csp_buffer.c#L56
- https://github.com/libcsp/libcsp/blob/libcsp-1/src/csp_conn.c#L98
- https://github.com/libcsp/libcsp/blob/libcsp-1/src/csp_port.c#L58

These memory allocation functions depend on the underlying OS to provide heap memory. For example, if you are using FreeRTOS, `pvPortMalloc() `must be functional. See https://github.com/libcsp/libcsp/blob/libcsp-1/src/arch/freertos/csp_malloc.c.

yashi, 2024-04-18:
@hernaangq FYI: https://www.freertos.org/a00111.html

hernaangq, 2024-04-18:
Hello @yashi thaks your the fast reply,

I am using the v1.6 indeed and doing it so with a static memory allocation, do you have any FreeRTOSCONFIG.h examples when using this version of csp??

Thank you.

moonlight83340, 2024-04-18:
> Hello @yashi thaks your the fast reply,
> 
> I am using the v1.6 indeed and doing it so with a static memory allocation, do you have any FreeRTOSCONFIG.h examples when using this version of csp??
> 
> Thank you.

Hi,

There is an example on the libcsp FreeRTOS test repository : 
https://github.com/libcsp/libcsp-freertos/blob/main/FreeRTOSConfig.h

But please note :  

> These definitions should be adjusted for your particular hardware and application requirements.

See also :
https://www.freertos.org/a00110.html

hernaangq, 2024-04-18:
Hello @moonlight83340 
> There is an example on the libcsp FreeRTOS test repository :
https://github.com/libcsp/libcsp-freertos/blob/main/FreeRTOSConfig.h

I have been using that one but it is from libcsp v2.0.
I would like a v1.6 one.

Thanks for your reply.

moonlight83340, 2024-04-18:
> Hello @moonlight83340
> 
> > There is an example on the libcsp FreeRTOS test repository :
> > https://github.com/libcsp/libcsp-freertos/blob/main/FreeRTOSConfig.h
> 
> I have been using that one but it is from libcsp v2.0. I would like a v1.6 one.
> 
> Also, in that example I can´t see the configAPPLICATION_ALLOCATED_HEAP param.
> 
> Thanks for your reply.

I'm not using FreeRTOS and it's seems a FreeRTOS configuration question.
Did you look at the https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/MemMang/heap_4.c ?
If you're using heap_4.c

You can check here more informations also :
https://www.freertos.org/a00110.html

hernaangq, 2024-04-18:
heap_4.c returns this error

```
#if ( configSUPPORT_DYNAMIC_ALLOCATION == 0 )
    #error This file must not be used if configSUPPORT_DYNAMIC_ALLOCATION is 0
#endif
```

which makes sense, because I want to use STATIC instead of DYNAMIC.

@moonlight83340 thanks for your answers.
@yashi do you use freertos?

moonlight83340, 2024-04-18:
> heap_4.c returns this error
>
> ```
> #if ( configSUPPORT_DYNAMIC_ALLOCATION == 0 )
>     #error This file must not be used if configSUPPORT_DYNAMIC_ALLOCATION is 0
> #endif
> ```
>
> which makes sense, because I [...]

That make sense. Maybe you should try to found a implementation which doesn't require configSUPPORT_DYNAMIC_ALLOCATION . And use static instead ?

I think heap_1,2,3,4,5 all use dynamic allocation 😓

yashi, 2024-04-19:
> I am using the v1.6 indeed and doing it so with a static memory allocation

Unfortunately, v1.6 supports it. It must have dynamic allocation support from OS.  Alas, `develop` branch still has some dynamic allocations, which we want to eliminate (#460).

> do you have any FreeRTOSCONFIG.h examples when using this version of csp??
> @yashi do you use freertos?

As I told you in #485, no.

yashi, 2024-04-20:
@hernaangq Reopen it if you have more question about it.  Or, open new one if you have some other issues.

closed 2024-04-20 by yashi

hernaangq, 2024-04-22:
hello, 
I have been debugging my code and let me tell you where is the error, maybe you can help me.
After debugging csp_buffer_init is okey but what sends the error is csp_conn_init, here I see lots of static allocations of memory.

thanks for all the help.
Hernan

hernaangq, 2024-04-22:
/libraries/libcsp_1_6_memoria_estatica/src/csp_conn.c:124: undefined reference to `csp_queue_create'

I also get this error.

hernaangq, 2024-04-22:
Okey I solved it initializing less conn_max in my csp conifg.

moonlight83340, 2024-04-22:
Did you debug your freeRTOS error ? If you are unsure of a something, share your code and I will take a look.