libcsp/libcsp #561: example of adding a route to rtable

hernaangq, 2024-04-24:
Good morning,
I would like to ask if there is any examples of how to add a route to rtable.
 I have already configure a can iface called "CAN_OBC_IF" and I want to associate it to a node 7 so I can send packets there.
But I don't know how.

thanks,
hernan

hernaangq, 2024-04-24:
``
int errorOK = csp_rtable_set(7, 0, &EJ, CSP_NO_VIA_ADDRESS);
``

EJ is my iface.

then I do:
``
csp_ping_noreply(7);
``

And can´t see anything in my CAN.

yashi, 2024-04-25:
I don't know how your network is configured. But as you wrote, `csp_rtable_set()` is what you need.  If you specify `CSP_NO_VIA_ADDRESS`, your packets will be directly to the address.

Try calling `csp_rbale_print()` to see how your interfaces are configured.

FYI: https://www.freecodecamp.org/news/how-to-ask-good-technical-questions/

yashi, 2024-04-30:
@hernaangq Let me know if you need more help by reopening this or new issue!  Thanks.

closed 2024-04-30 by yashi