libcsp/libcsp #565: Dynamic Allocation in SFP

jeffstjean, 2024-04-27:
I noticed that csp_sfp_send_own_memcpy has an option for a user-specified memcpy function but it seems like csp_sfp_recv_fp requires the use of malloc. 

Are there plans to add the ability to either:
1. Pass a user-specified allocator
2. Pass a fixed size buffer?

I would like to use SFP rather than implementing my own protocol but can’t use any dynamic memory.

yashi, 2024-04-28:
Yes, we do have a plan.  It's #460.
As always, any help would be appreciated.  Thanks.

dimitrovand, 2024-05-29:
Hello, is anyone working on removing `malloc` usage from `sfp`? If not - is it better to implement something more generic? Do you think the following example implementation matches the desired outcome:
```
/**
 * Structure to be passed as an input parameter to csp_sfp_send(...).
 * This structure encapsulates user-defined data and a function for reading data from storage.
 */
typedef struct {
    /**
     * User-defined data. The SFP does not interpret or manipulate this data.
     * It can be any user-defined data such as a file handle, a raw buffer pointer, or any other 
     * relevant information required for the implementation.
     */
    void* user_data;

    /** 
     * User-defined function for reading data from any type of storage.
     * 
     * @param[in] user_data Pointer to user-specific data required for the read operation.
     * @param[out] data Pointer to the buffer where the read data should be copied.
     * @param[in] offset The offset in the storage from where the data should be read.
     * @param[in] size The number of bytes to read from the storage.
     * 
     * @return #CSP_ERR_NONE on successful read, otherwise an error code indicating the failure.
     */
    int (*read)(void* user_data, uint8_t* data, uint32_t offset, uint32_t size);
} csp_sfp_read_t;

/**
 * Structure to be passed as an input parameter to csp_sfp_recv(...).
 * This structure encapsulates user-defined data and a function for writing data to storage.
 */
typedef struct {
    /**
     * User-defined data. The SFP does not interpret or manipulate this data.
     * It can be any user-defined data such as a file handle, a raw buffer pointer, or any other 
     * relevant information required for the implementation.
     */
    void* user_data;

    /** 
     * User-defined function for writing data to any type of storage.
     * 
     * @param[in] user_data Pointer to user-specific data required for the write operation.
     * @param[in] data Pointer to the buffer containing the data to be written.
     * @param[in] offset The offset in the storage where the data should be written.
     * @param[in] size The number of bytes to write to the storage.
     * 
     * @return #CSP_ERR_NONE on successful write, otherwise an error code indicating the failure.
     */
    int (*write)(void* user_data, uint8_t* data, uint32_t offset, uint32_t size);
} csp_sfp_write_t;
```