libcsp/libcsp #570: Enable printout when using libcsp as a submodule

robamu, 2024-05-16:
I could not really figure this out quickly from the docs and I think this is a bit confusing. I am using CMake to build my application and I have added `libcsp` as a git submodule, and linked against it. How exactly do I configure libcsp to print everything? Do I always have to manually edit the `csp_autoconfig.h` file inside my build directory? I tried `set (CSP_PRINT_STDIO 1)` in my `CMakeLists.txt` and
`-DCSP_PRINT_STDIO=1` in my CMake build generation command, but those do not have any effect probably because this is set using `set` instead of `option`.

Some documentation would be good here as well. If the STDIO print is OFF and I try to run the example.. Nothing happens superficially because nothing is printed.

robamu, 2024-05-16:
Well, never mind, I was using v2.0. This was fixed/changed in v2.1. However, I am leaving this open for the call to improve the documentation :-)

yashi, 2024-05-20:
The problem, aside from the documentation, is that we initially integrated CMake solely for Zephyr RTOS, which is the case for v2.0.

However, since then, people have started using it for Linux, and now FreeRTOS users want to use it as well, as FreeRTOS proper appears to support CMake.

PR #574 (2024-05-21): INSTALL: Add note about preliminary CMake support in v2.0
In version 2.0, CMake support is preliminary. This change ensures users are aware of this limitation.

This fixes #570.
closed 2024-05-24 by yashi