libcsp/libcsp #572: Building the documentation

robamu, 2024-05-17:
I am trying to build the documentation for the most recent develop branch and I have issue doing that.
I think a good thing would be to document how the docs are built.

I did the following:

```sh
cd doc
python3 -m venv venv
. venv/bin/activate
pip install -r requirements.txt
mkdir build && cd build
cmake ..
cmake --build . -j 10
```

I am not sure if all of this is even required. Why is there another CMake wrapper necessary around the Sphinx workflow? I am used to just running `make html` in regular Sphinx projects.

I then get this error:

```sh
libcsp/doc/build on  develop [$!?] via △ v3.22.1 via 🐍 v3.10.12 (venv) took 2s
❯ cmake --build . -j 10
[100%] Building DOCS for lib CSP
Running Sphinx v5.3.0
myst v1.0.0: MdParserConfig(commonmark_only=False, gfm_only=False, enable_extensions=set(), disable_syntax=[], all_links_external=False, url_schemes=('http', 'https', 'mailto', 'ftp'), ref_domains=None, fence_as_directive=set(), number_code_blocks=[], title_to_header=False, heading_anchors=0, heading_slug_func=None, html_meta={}, footnote_transition=True, words_per_minute=200, substitutions={}, linkify_fuzzy_links=True, dmath_allow_labels=True, dmath_allow_space=True, dmath_allow_digits=True, dmath_double_inline=False, update_mathjax=True, mathjax_classes='tex2jax_process|mathjax_process|math|output_area', enable_checkboxes=False, suppress_warnings=[], highlight_code_blocks=True)
building [mo]: targets for 0 po files that are specified
building [html]: 53 source files given on command line
updating environment: [new config] 85 added, 0 changed, 0 removed
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of csp_sfp.h:7: ERROR: Unexpected indentation.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of csp_sfp.h:4: WARNING: Block quote ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of csp_sfp.h:11: ERROR: Unexpected indentation.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of csp_sfp.h:8: WARNING: Block quote ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of csp_sfp.h:7: ERROR: Unexpected indentation.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of csp_sfp.h:4: WARNING: Block quote ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of drivers/can_socketcan.h:8: ERROR: Unexpected indentation.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of drivers/can_socketcan.h:5: WARNING: Block quote ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of drivers/can_socketcan.h:7: WARNING: Definition list ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of drivers/can_socketcan.h:9: ERROR: Unexpected indentation.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of drivers/can_socketcan.h:6: WARNING: Block quote ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of drivers/can_socketcan.h:8: WARNING: Definition list ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of drivers/eth_linux.h:8: ERROR: Unexpected indentation.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of drivers/eth_linux.h:5: WARNING: Block quote ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of interfaces/csp_if_can.h:1: WARNING: Definition list ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of interfaces/csp_if_i2c.h:1: WARNING: Definition list ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of interfaces/csp_if_kiss.h:1: WARNING: Definition list ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of interfaces/csp_if_zmqhub.h:3: WARNING: Definition list ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of interfaces/csp_if_zmqhub.h:5: WARNING: Definition list ends without a blank line; unexpected unindent.
/home/rmueller/Software/csp-client-sec-challenge/libcsp/doc/venv/lib/python3.10/site-packages/sphinx_c_autodoc/loader.py:docstring of interfaces/csp_if_zmqhub.h:8: WARNING: Definition list ends without a blank line; unexpected unindent.

Extension error:
Handler <function include_readme_file at 0x70136516c8b0> for event 'source-read' threw an exception (exception: [Errno 2] No such file or directory: '../README.md')
gmake[2]: *** [CMakeFiles/docs.dir/build.make:73: html/index.html] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/docs.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2

```

robamu, 2024-05-17:
Also it might be worth a thought to switch from `sphinx_c_autodoc` to [`breathe`](https://breathe.readthedocs.io/en/latest/).
This tool does not require a `clang` installation as far as I know, so this would simplify building the documentation.

yashi, 2024-05-20:
Ah, I totally forgot about this.

As noted in this commit https://github.com/libcsp/libcsp/commit/db083f6eddd89355922e480b411470ad8cf6505a, we have a problem. You can't build the documentation in doc/; it must be done in the root directory.

If you know a better way to fix it, it would be greatly appreciated.

robamu, 2024-05-20:
For the issue of allowing building inside `doc` ? One way would be to try to find the `README.md` file in the upper source directory and the current source directory. Probably wouldn't be too compliated in Python.

yashi, 2024-05-21:
Searching `README.md` for both dir won't work because you can specify your build dir at an arbitrary place, which is perfectly valid with CMake.

It seems the Sphinix instance has [srcdir](https://www.sphinx-doc.org/en/master/extdev/appapi.html#sphinx.application.Sphinx.srcdir).  I've added a fix to the PR.

closed 2024-05-21 by yashi

yashi, 2024-05-21:
This means you can run a command like the following from anywhere.

```
cmake -S ~/src/libcsp/doc -B /tmp/libcsp-doc && ninja -C /tmp/libcsp-doc
```

PR #575 (2024-05-21): doc: Add build-doc
This PR has two commits.

1. Add build instruction
2. Remove restriction where to exec by fixing relative README.md path calc.

This fixes #572.