libcsp/libcsp #573: Zephyr with CSP CAN

Ed4129, 2024-05-19:
Hi there,

I am trying to set up CSP CAN to work with Zephyr on an NXP development board. If there is relevant documentation about this, could somebody point me in the right direction, please? 

After setting up Zephyr and successfully getting it to work with a few projects, I tried to apply the same to CSP CAN using the following commands.

At relative path libcsp/zephyr/samples/server-client: 
west build -b MIMXRT1050_evk ../server-client

I obtained the following errors. Are there any preprocessing and pre-build steps necessary before building it with Zephyr? There are certain Preprocessor variables such as LIBCSP, CONFIG_CSP_HAVE_CAN that are undefined - I suppose those would have to be manually defined before the example will work? Please advise, thanks. Ed

Loading Zephyr default modules (Zephyr base).
-- Application: /home/ed/zephyrproject/zephyr/libcsp/zephyr/samples/server-client
-- CMake version: 3.29.3
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /home/ed/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/ed/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: mimxrt1050_evk, Revision: hyperflash, qualifiers: mimxrt1052
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/home/ed/zephyr-sdk-0.16.5-1)
-- Found toolchain: zephyr 0.16.5 (/home/ed/zephyr-sdk-0.16.5-1)
-- Found Dtc: /home/ed/zephyr-sdk-0.16.5-1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/ed/zephyrproject/zephyr/boards/nxp/mimxrt1050_evk/mimxrt1050_evk.dts
-- Found devicetree overlay: /home/ed/zephyrproject/zephyr/boards/nxp/mimxrt1050_evk/mimxrt1050_evk_mimxrt1052_hyperflash.overlay
-- Generated zephyr.dts: /home/ed/zephyrproject/zephyr/libcsp/zephyr/samples/server-client/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/ed/zephyrproject/zephyr/libcsp/zephyr/samples/server-client/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/ed/zephyrproject/zephyr/libcsp/zephyr/samples/server-client/build/zephyr/dts.cmake

/home/ed/zephyrproject/zephyr/libcsp/zephyr/samples/server-client/prj.conf:1: warning: attempt to assign the value 'y' to the undefined symbol LIBCSP

/home/ed/zephyrproject/zephyr/libcsp/zephyr/samples/server-client/prj.conf:2: warning: attempt to assign the value 'n' to the undefined symbol LIBCSP_HAVE_STDIO

/home/ed/zephyrproject/zephyr/libcsp/zephyr/samples/server-client/prj.conf:3: warning: attempt to assign the value 'n' to the undefined symbol CSP_USE_RTABLE

error: Aborting due to Kconfig warnings

Parsing /home/ed/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/ed/zephyrproject/zephyr/boards/nxp/mimxrt1050_evk/mimxrt1050_evk_defconfig'
Merged configuration '/home/ed/zephyrproject/zephyr/libcsp/zephyr/samples/server-client/prj.conf'
CMake Error at /home/ed/zephyrproject/zephyr/cmake/modules/kconfig.cmake:391 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/ed/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /home/ed/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/ed/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/ed/zephyrproject/zephyr/libcsp/contrib/zephyr/samples/server-client/build -GNinja -DBOARD=mimxrt1050_evk -S/home/ed/zephyrproject/zephyr/libcsp/contrib/zephyr/samples/server-client

yashi, 2024-05-20:
[This](https://github.com/libcsp/libcsp-zephyr/blob/main/west.yml) is how you write your west.yml.

```yaml
manifest:

  projects:
    - name: zephyr
      url: https://github.com/zephyrproject-rtos/zephyr
      revision: main
      path: zephyr
      west-commands: scripts/west-commands.yml
      import: true

    - name: libcsp
      url: https://github.com/libcsp/libcsp
      revision: develop
      path: modules/lib/libcsp
```

Ed4129, 2024-05-23:
Thank you! Managed to get it working after editing the yml file.