libcsp/libcsp #576: Question: Does libCSP add any header or bits AFTER the message?

hernaangq, 2024-05-22:
Good morning,

I am sending CSP messages from a node to another one via CAN and printing them. 
I can see there are some bits after the message, is this how it is suppose to work?
The message I am sending is just the word "hello".

[image]

moonlight83340, 2024-05-23:
Hi,
what version and option are you using ?

hernaangq, 2024-05-23:
Hello, 
The client is using 1.4 and server 1.6. 
what options do you mean?

moonlight83340, 2024-05-23:
Sorry,
I meant what flag are you using ?
RDP, CRC32 for example.

hernaangq, 2024-05-23:
/* WARNING! All changes made to this file will be lost! */

#ifndef W_INCLUDE_CSP_CSP_AUTOCONFIG_H_WAF
#define W_INCLUDE_CSP_CSP_AUTOCONFIG_H_WAF

#define GIT_REV "7d03adb"
#define CSP_FREERTOS 1
/* #undef CSP_POSIX */
/* #undef CSP_WINDOWS */
/* #undef CSP_MACOSX */
#define CSP_DEBUG 1
#define CSP_DEBUG_TIMESTAMP 0
#define CSP_USE_RDP 1
#define CSP_USE_RDP_FAST_CLOSE 0
#define CSP_USE_CRC32 1
#define CSP_USE_HMAC 0
#define CSP_USE_XTEA 0
#define CSP_USE_PROMISC 0
#define CSP_USE_QOS 0
#define CSP_USE_DEDUP 0
#define CSP_USE_EXTERNAL_DEBUG 1
#define CSP_LOG_LEVEL_DEBUG 1
#define CSP_LOG_LEVEL_INFO 1
#define CSP_LOG_LEVEL_WARN 1
#define CSP_LOG_LEVEL_ERROR 1
/* #undef CSP_LITTLE_ENDIAN */
#define CSP_BIG_ENDIAN 1
#define LIBCSP_VERSION "1.6"

#endif /* W_INCLUDE_CSP_CSP_AUTOCONFIG_H_WAF */

I am using those.

moonlight83340, 2024-05-23:
Sorry again, I meant thoses flags :
https://github.com/libcsp/libcsp/blob/447cc38f2106a15290358dc8c128ad553a415568/examples/csp_client.c#L252

In csp_connect()``

hernaangq, 2024-05-23:
no problem, thanks for taking the time to answer my question.

In the client side:
csp_conn_t * conn = csp_connect(CSP_PRIO_NORM, node, 5, 0, 0);

In the server side:

hernaangq, 2024-05-23:
this might help, 
Now I am sending "hello cube" and with a CAN byte reader I am getting the following results:

[image]

yashi, 2024-05-24:
`B9 E8 96 4C` must be the CRC32. See https://crccalc.com/?crc=hello cube&method=CRC-32C&datatype=ascii&outtype=0

hernaangq, 2024-05-24:
thank you so much, that was my first thought. 
I guess csp_read is the function that get rid of this CRC right?

moonlight83340, 2024-05-24:
> thank you so much, that was my first thought. I guess csp_read is the function that get rid of this CRC right?

In fact CRC is strip on the security check : 
https://github.com/libcsp/libcsp/blob/v1.4/src/csp_crc32.c#L124
https://github.com/libcsp/libcsp/blob/v1.4/src/csp_route.c#L89C7-L89C23
https://github.com/libcsp/libcsp/blob/v1.4/src/csp_route.c#L129C5-L129C19
It's done when a packet is received on the route.

hernaangq, 2024-05-24:
```
    while (1)
    {
       
        conn = csp_accept(sock, 10000);

        if (conn == NULL)
        {
            continue;
        }

        while ((packet = csp_read(conn, 0)) != NULL)
        {
            if (packet == NULL)
                break;

            debug_printf_safe(INFO, "CSP PACKET RECEIVED");
            csp_buffer_free(packet);
        }

    
        csp_close(conn);
        
    }
```

So this code should strip de CRC automatically or should I add any code?

moonlight83340, 2024-05-24:
> ```
>     while (1)
>     {
>
>         conn = csp_accept(sock, 10000);
>
>         if (conn == NULL)
>         {
>             continue;
>         }
>
>         while ((packet = csp_read(conn, 0)) != [...]

I'm curious about your client code, if you mind ?
I think you should use CRC32 option flag to be automatically strip.

hernaangq, 2024-05-24:
```
/* Prepare data */
	csp_packet_t * packet = csp_buffer_get(5);
	if (packet == NULL)
		return;

	/* Open connection */
	csp_conn_t * conn = csp_connect(CSP_PRIO_NORM, node, 5, 0, 0);
	if (conn == NULL) {
		csp_buffer_free(packet);
		return;
	}

	//packet->data[0] = 0x77;
	//packet->data[1] = 0x6F;
	//packet->data[2] = 0x72;
	//packet->data[3] = 0x6c;
	//packet->data[4] = 0x64;

	packet->data[0] = 0x68;
	packet->data[1] = 0x65;
	packet->data[2] = 0x6c;
	packet->data[3] = 0x6c;
	packet->data[4] = 0x6f;
	packet->data[5] = 0x20;
	packet->data[6] = 0x63;
	packet->data[7] = 0x75;
	packet->data[8] = 0x62;
	packet->data[9] = 0x65;

	packet->length = 10;

	//printf(" %u.\r\n", packet->data);
	printf("Sending: %s\n", (char *)packet->data);

	/* Try to send frame */
	if (!csp_send(conn, packet, 0))
		csp_buffer_free(packet);

	csp_close(conn);

```

Here you have it, how do I add that CRC32 option flag to be automatically strip?

moonlight83340, 2024-05-24:
> ```
> /* Prepare data */
> 	csp_packet_t * packet = csp_buffer_get(5);
> 	if (packet == NULL)
> 		return;
>
> 	/* Open connection */
> 	csp_conn_t * conn = csp_connect(CSP_PRIO_NORM, node, 5, 0, 0);
> 	if (conn == [...]

Use this flag : `CSP_O_CRC32` instead of your last 0.
https://github.com/libcsp/libcsp/blob/v1.4/include/csp/csp_types.h#L159

hernaangq, 2024-05-27:
@moonlight83340 Thank you so much, that was it.

closed 2024-05-27 by hernaangq