libcsp/libcsp #577: Getting CSPCAN examples to work.

Ed4129, 2024-05-23:
Hi, 

I am trying to get the following example to work on my NXP board as a CSP CAN client and a Raspberry Pi node as a CSP CAN server. The only change I made is address = 0, and server_address = 27. May I know what changes I ought to make to get the example to work? 

Thanks,
Ed 

csp_server_client_zephyr.c
main.c

I am using python_bindings_example_client.py on the Raspberry Pi Client side. I am assuming that the server address is set to 27. 

libcsp.init(27, "test_service", "bindings", "1.2.3", 10, 300)
libcsp.zmqhub_init(27, "localhost")
libcsp.rtable_set(0, 0, "ZMQHUB")
libcsp.route_start_task()

However, there doesn't seem to be any ping transactions coming through. The log on the Client side is as follows and no ping tx is followed.

*** Booting Zephyr OS build v3.6.0-4321-gec0e2d630868 ***
[00:00:00.001,000] <inf> csp_sample_server_client: Initialising CSP
[00:00:00.002,000] <inf> csp_sample_server_client: CAN interface added successfully.
[00:00:00.002,000] <inf> csp_sample_server_client: Connection table
[00:00:00.002,000] <inf> csp_sample_server_client: Interfaces
[00:00:00.002,000] <inf> csp_sample_server_client: Client started.
[00:00:00.002,000] <inf> csp_sample_server_client: Client task started

The log tx is as follows.

LD_LIBRARY_PATH=build PYTHONPATH=build python3 examples/python_bindings_example_server.py
Hostname: test_service
Model:    bindings
Revision: 1.2.3
Routes:
27/5 LOOP
0/0 ZMQHUB

yashi, 2024-05-24:
> The only change I made is address = 0

Address 0 is a special reserved address for the local.  You can't send any packet out to `0`.

yashi, 2024-06-05:
@Ed4129 Please reopen this or open a new one if you have more questions. Thanks.

closed 2024-06-05 by yashi