libcsp/libcsp #579: Providing a Rust bindings for the CSP library in Rust

robamu, 2024-05-28:
Hi,

Rust is starting to get more common in the embedded/space domain as well, so I think it would be nice to provide binding bindings and a nice API on top of the bindings to allow integrating the CSP library into a Rust project.

Is there any ongoing effort by any csp developers for this? I was always interested in learning how to do this and I think this would be a really nice addition. I might have a shot at this, and then try to re-implement some of the examples provided in Rust. What do you think?

I know there is https://github.com/TolimanSpace/libcsp-rs, but that seems to point to an older libcsp version, but that seems to use an older version.

Kind Regards
Robin

yashi, 2024-05-31:
> Is there any ongoing effort by any csp developers for this?

Not that I know of.

> I might have a shot at this, and then try to re-implement some of the examples provided in Rust. What do you think?

I love Rust and use it for non-embedded systems. I know Rust is not as widely supported as C for MCUs. However, that shouldn't stop us from writing a wrapper for it. Unfortunately, we don't have the manpower to maintain it. If you or someone else steps up to write and _maintain_ it, I'd be happy to accept the PR.

robamu, 2024-05-31:
I was actually able to make good progress. Quite some bit of engineering necessary to make everything work smoothly, but I am on a good way to make the server/client example work in Rust.

Is there any plan on releasing v2.1 soon? If I release anything on crates.io, I'd prefer to have it be based on a released version ideally.

robamu, 2024-06-01:
I published a first usable version.
You can find the repository here: https://github.com/us-irs/libcsp-rust

I fully implemented the client/server example code. This allows clean interoperability with libcsp in regular Rust code
on POSIX systems, which is already worth a lot in my opinion.
I am particular happy with the API I was able to create for the client/server example. With Rust, I was able to create guarded APIs on top of the raw C FFI API which take care of freeing buffers and/or closing connections automatically.

See here: https://github.com/us-irs/libcsp-rust/blob/main/examples/src/main.rs

It might be a better idea to keep these repositories separate for now..  Mentioning somewhere in the documentation that
libcsp can be used in Rust code and then specifying and cross-linking the repository might be
a good first step.

Some ramblings on how this work could be further enhanced and improved:

Right now, the `libcsp-sys` crate does not build the C library directly. It would be nice if this is done and I would also like th integrate `pkg-config` in some shape or form so a Rust app on Linux can just use the system-installed libcsp. I think building libcsp manually directly is still an important use-case and that is how it is done right now, so I'd probably have to add some config based build of the library.

The most challenging part now is probably to also make this work for other RTOSes. I think the integration with C RTOSes might be a real challenge, I'd have to get familiar with how  to handle transitive dependencies (native Rust code with depends on both libcsp and FreeRTOS,  with libcsp depending on FreeRTOS as well). I think what might be more interesting might be the integration of libcsp into Rust RTOSes like RTIC or embassy. This would probably require some Rust to C interoperability.

yashi, 2024-06-02:
> Is there any plan on releasing v2.1 soon?

We haven't decided yet. But if other parts of ecosystem need a new libcsp version, I'd say we should release more often.

> It might be a better idea to keep these repositories separate for now

If that works for you, that's great. (less work for me ;-)

I thought it might be easier to update the APIs if you integrate with libcsp proper since we still don't have a stable API or policy for it.

> How easy or difficult is it to write the arch specific part for libcsp? 

See 630c1e96925426e8013e569cc9646e51603e296a and the following commits.

yashi, 2024-06-05:
@robamu Please reopen it if you have more question.  Thanks.

closed 2024-06-05 by yashi