libcsp/libcsp #596: CRC32 verification depend on a non protect bit that could be corrupt

moonlight83340, 2024-06-24:
**Brief**
When using LIBCSP with the CSP_O_CRC32 option, we should throw away corrupted messages. However, if the option bit is corrupted, I think the packet can bypass the security route because it wouldn't be verified.

**Current understanding**
I found in the source code how the packet can skip the security path with a bit change in the packet header:
https://github.com/libcsp/libcsp/blob/develop/src/csp_route.c#L63

**Expected behavior**
If a message is corrupted, it should be thrown away.

**Impact**
A message can pass the CRC32 security with only a bit change, which shouldn't be the case.

**Correction**
I don't know yet how to avoid this situation, but I wanted to share it with others.

dimitrovand, 2024-07-09:
@moonlight83340  Good point. The only proper solution might be to always add and check the CRC of outgoing and incoming packets. If this is not done, the responsibility lies with the underlying driver implementation. Of course, this is not an issue when CAN bus is being used.

moonlight83340, 2024-07-09:
> Although, it seems like you can avoid corrupted packets by simply adding a CRC flag to your socket options. The router checks the socket/connection options against the incoming packet options and if they don't match the packet seems to be dropped. The check is performed here https://github.com/libcsp/libcsp/blob/develop/src/csp_route.c#L69 . So if the packet is with CRC flag set to 0 but the socket accepts only packets with CRC flag set to 1 -> packet is dropped.

I'm not sure but based on what I have understood with the source code. The option can be set on client side but not on server side ? 
Am I wrong ?

dimitrovand, 2024-07-09:
Every time you call `csp_accept() `on server side, and it succeeds, a connection has been created by `csp_route_work()` again on the server side, e.g.:

 https://github.com/libcsp/libcsp/blob/develop/src/csp_route.c#L250 this is where the new connection inherits the socket options
https://github.com/libcsp/libcsp/blob/develop/src/csp_route.c#L282 this is where the new connection is queued in your socket queue (at this point `csp_accept() `is unblocked if your are waiting on it).

To test it - just send a packet without CRC to a destination socket with CRC OPT enabled and on your server side you can put print statements inside `csp_route_work() `to verify whether or not the packet gets rejected.

moonlight83340, 2024-07-09:
> Every time you call `csp_accept() `on server side, and it succeeds, a connection has been created by `csp_route_work()` again on the server side, e.g.: [...]

I meant where do we init the  CRC OPT on server side ? I'm not sure.

dimitrovand, 2024-07-09:
I don't think there is an existing example, but you can add this `sock.opts = CSP_SO_CRC32REQ; ` here: https://github.com/libcsp/libcsp/blob/develop/examples/csp_server.c#L43 and run the example. There are no functions for modifying the socket options, as far as I am aware, so you have to do it with direct modification of .opts.

moonlight83340, 2024-07-09:
Thank you, I wasn't sure about that, I will test with the option.
It would be nice to have a function or to be able to configure the socket option when creating the server socket.

closed 2024-07-11 by moonlight83340