libcsp/libcsp #603: CSP RDP Queue is not thread-safe

yashi, 2024-07-04:
Currently CSP RDP queue manipulation code (`csp_rdp_queue.c`), at least, is not thread safe.  It's impossible to use the code from multiple threads.

dimitrovand, 2024-09-06:
I can confirm the issue is present and reproducible. It's difficult to think of a clean solution that doesn't involve reintroducing mutexes...

yashi, 2024-09-06:
If we were to implement this from scratch, I would define a `struct csp_rdp_queue`. However, the current implementation uses implicit queues.

The problematic part is the dequeue -> enqueue section in the flush and get functions. Therefore, the simplest solution would be to add a lock around these sections.

The nobrainer approach would be to just add locks to the public functions.

johandc, 2024-09-18:
I think we whould consider using a lighter weight system and then go back to per-connection.
The issue with the old per-connection queues, was that they had to be statically pre-allocated.
If we would use linked-lists, with a simple atomic.h lock for pertubations, the memory size is extremely small. And there will be no additional memory required to store queues, and we dont invoke the larger queue api.

The queue api was specifically chosen for its thread safety, but with the way we have to take out, and insert to scan through a list ruins that anyways.

When the compiler supports atomic operations a mutex lock is extremely simple to add on the linked list.