libcsp/libcsp #605: How to use csp_server_client.c from your example in libcsp-2-0

poomravint, 2024-07-05:
1. I don't understand router_start(); what should i do with this function then i comment this function
2. server_start(); and client_start() i have to use pthread for call this function right ?
When i run code client can't ping to server i got this :
[image]

moonlight83340, 2024-07-05:
Hi,
Did you check the documentation on how to run the [example code ](https://libcsp.github.io/libcsp/example.html)?

poomravint, 2024-07-05:
Sure if run like this i can run and it work
[image]
but i try to run csp_server_client.c then client can't ping to server
actually i have try to coding CSP for 3 weeks but now i still can't T-T
can you teach me pls or guide me pls

> Hi, Did you check the documentation on how to run the [example code ](https://libcsp.github.io/libcsp/example.html)?

moonlight83340, 2024-07-05:
I can try to help you of course !
Can you tell me every commands you used and if you have done modifications on the example ?

poomravint, 2024-07-06:
> I can try to help you of course ! Can you tell me every commands you used and if you have done modifications on the example ?

Thank you very much i will show my code on this monday my code is in Raspi and Raspi is in office and now this is days off 
Thank you very much again for your help.

poomravint, 2024-07-08:
I add pthread library for use thread
[image]
then i make function to call server() and client() by thread 
[image]
then i comment router_star() because i don't know what should i do about this function
[image]
then i comment ".checkparity = 0" in csp_usart_conf_t conf
[image]
then i use pthread to start server_thread and client_thread
[image]
that's all i did fix it. I use example from this : https://github.com/libcsp/libcsp/blob/libcsp-2-0/examples/csp_server_client.c

moonlight83340, 2024-07-08:
So you manage to fix it with that ? I will take a look.

poomravint, 2024-07-08:
> So you manage to fix it with that ? I will take a look.

Yes or Do you wanna watch my full code

moonlight83340, 2024-07-08:
> > So you manage to fix it with that ? I will take a look.
> 
> Yes or Do you wanna watch my full code but actually i have concern about router_start() becuase i commended this function because i don't know what should i do. that's why my code is not working

If you can send me your full that would be great. (I'm not sure at the time why you have to comment router_start())
Before I ask you to tell me what command do you used. You really just try to execute csp_server_client.c ? And from start client and server wouldn't worked ?
What's your OS ?

poomravint, 2024-07-08:
> If you can send me your full that would be great. (I'm not sure at the time why you have to comment router_start()) Before I ask you to tell me what command do you used. You really just try to [...]

Sure how can i send a full code on this comment or any contact ?
I don't understand what is mean of command , command to compile and run or somethings ?
Yes i just try to run csp_server_client.c then i got this : 
[image]
start client and server call client() and server() that i told you it's didn't work i mena client can't ping to server and can't connected to server.
I use RaspiOS.

moonlight83340, 2024-07-08:
> Sure how can i send a full code on this comment or any contact ? I don't understand what is mean of command , command to compile and run or somethings ? Yes i just try to run csp_server_client.c [...]

You can zip it and post it here or gits.
Sorry by command I mean from compilation to execution.
Thank you for your time !

poomravint, 2024-07-08:
> You can zip it and post it here or gits. Sorry by command I mean from compilation to execution. Thank you for your time !

This is my zip : 
[csp_server_client.zip](https://github.com/user-attachments/files/16122407/csp_server_client.zip)

I compile by gcc csp_server_client.c  -lcsp -lpthread -o test2 ../libcsp/src/csp_rtable_cidr.c ../libcsp/src/csp_rtable_stdio.c
and run by ./test2

moonlight83340, 2024-07-08:
Did you first try without modification the csp_server_client.c ?
`./build/examples/csp_server_client`
When I run the soft with this command everything seems fine but I'm on an ubuntu.

poomravint, 2024-07-08:
> Did you first try without modification the csp_server_client.c ? `./build/examples/csp_server_client` When I run the soft with this command everything seems fine but I'm on an ubuntu.

if run with this ./build/examples/csp_server_client i can run then client can ping and send message to server
but why my code client can't ping and send message to server

poomravint, 2024-07-08:
> > Did you first try without modification the csp_server_client.c ? `./build/examples/csp_server_client` When I run the soft with this command everything seems fine but I'm on an ubuntu.
> 
> if run with this ./build/examples/csp_server_client i can run then client can ping and send message to server but why my code client can't ping and send message to server or can you teach me how to use CSP i think i don't understand how to use CSP

poomravint, 2024-07-08:
> > > Did you first try without modification the csp_server_client.c ? `./build/examples/csp_server_client` When I run the soft with this command everything seems fine but I'm on an ubuntu.
> > 
> > 
> > if run with this ./build/examples/csp_server_client i can run then client can ping and send message to server but why my code client can't ping and send message to server
> 
> I'm sorry, I'm not sure I understand well. You're trying to run csp_server_client only ? or did you try to connect another client on the csp_server_client ?
> 
> Maybe I didn't understood clearly what you're doing.

Yes now i try to run only csp_server_client 
I just want to make receiver.c and sender.c
I use receiver.c be a server to receiver message from sender.c (No interface)
and I use sender.c be a client  to send message to receiver.c (No interface)
Can you coding CSP to send or receiver messages ?

moonlight83340, 2024-07-08:
Okay, I finally understood sorry.
You could check the current develop branch, examples has been modify to separate client and server.
https://github.com/libcsp/libcsp/tree/develop/examples
I think that what you want to understand.

poomravint, 2024-07-08:
> Okay, I finally understood sorry. You could check the current develop branch, examples has been modify to separate client and server. https://github.com/libcsp/libcsp/tree/develop/examples I think that what you want to understand.

Can you use code from examples then it's work right ?

moonlight83340, 2024-07-08:
> > Okay, I finally understood sorry. You could check the current develop branch, examples has been modify to separate client and server. https://github.com/libcsp/libcsp/tree/develop/examples I think that what you want to understand.
> 
> Can you use code from examples then it's work right ?

It would need some modification for sure. But it can help you to have a minimal server and client.

poomravint, 2024-07-08:
> It would need some modification for sure. But it can help you to have a minimal server and client.

Yeah i know example code need some modification .I have tried to modification for many time then I can't use CSP
[image]
I don't understand about this function what I have to do about this and I don't know how to set address for a server

moonlight83340, 2024-07-09:
Sorry for long time,
`router_start()` will call csp_route_work
https://github.com/libcsp/libcsp/blob/develop/src/csp_route.c#L109
It's an useful part of libcsp :sweat_smile: 

Because I usually not running libcsp with no interface, I'm unsure if you can run a client against a server from different file with no interface.

poomravint, 2024-07-09:
> csp_route_work

> Sorry for long time, `router_start()` will call csp_route_work https://github.com/libcsp/libcsp/blob/develop/src/csp_route.c#L109 It's an useful part of libcsp 😅
> 
> Because I usually not running libcsp with no interface, I'm unsure if you can run a client against a server from different file with no interface.

Thanks I will try it.
You told you usually running by interface right ?
What's interface do you use ?

moonlight83340, 2024-07-09:
I used KISS / CAN and zmq but only on local test.

poomravint, 2024-07-09:
> I used KISS / CAN and zmq but only on local test.

Can I watch your code or Can you teach me ?
Actually I try to send data by UART between 2 Raspi by CSP but I don't understand how to config in code

moonlight83340, 2024-07-09:
Base on libcsp example I have something like that.
I will run csp_client and csp_server from different program.
[csp_client-csp_server-example.zip](https://github.com/user-attachments/files/16141622/csp_client-csp_server-example.zip)
I haven't used UART but I can check if I have some example to work on.
Yes csp_route_work should be run on another thread.
I hope you can understand my code, I can explain to you.

poomravint, 2024-07-10:
> Base on libcsp example I have something like that. I will run csp_client and csp_server from different program. [...]

Thanks a lot I will take a look

poomravint, 2024-07-10:
> Base on libcsp example I have something like that. I will run csp_client and csp_server from different program. [...]

from your code Can you run csp_client.c and csp_server.c to communicate ?
I try to run csp_client.c and csp_receiver.c but client can't ping or send data to server

moonlight83340, 2024-07-10:
> from your code Can you run csp_client.c and csp_server.c to communicate ? I try to run csp_client.c and csp_receiver.c but client can't ping or send data to server

I have try using zmq and with zmq I can.

poomravint, 2024-07-10:
> > > Base on libcsp example I have something like that. I will run csp_client and csp_server from different program. [csp_client-csp_server-example.zip](https://github.com/user-attachments/files/16141622/csp_client-csp_server-example.zip) I haven't used UART but I can check if I have some example to work on. Yes csp_route_work should be run on another thread. I hope you can understand my code, I can explain to you.
> > 
> > 
> > from your code Can you run csp_client.c and csp_server.c to communicate ? I try to run csp_client.c and csp_receiver.c but client can't ping or send data to server
> 
> I have try using zmq and with zmq I can. (I can't without interface)

Can i see your code that you use interface please ?

moonlight83340, 2024-07-10:
On the source code in the zip you can see : 

`/* Add interface(s) */`
From that commentary you will have the code that init the interface you choose to execute.
So in my case it will use `csp_zmqhub_init`

To run I used the same command as in the documentation for running zmq in 3 differents terminal : 
https://libcsp.github.io/libcsp/example.html

To run proxy
`libcsp$ ./build/examples/zmqproxy`
To run server
`libcsp$ ./build/examples/csp_server -z localhost -a 2`
To run client
`libcsp$ ./build/examples/csp_client -z localhost -a 3 -C 2`

poomravint, 2024-07-10:
> On the source code in the zip you can see :
>
> `/* Add interface(s) */` From that commentary you will have the code that init the interface you choose to execute. So in my case it will use [...]

 Thanks I will try it later now I am coding 2 process to communicate but no interface

yashi, 2024-07-11:
Please reopne it or open new one if you have more questions. Thanks.

closed 2024-07-11 by yashi