libcsp/libcsp #606: Question: Libcsp ports

AngelPerezM, 2024-07-06:
Hi all:

First of all, this is not an issue, it's a question related to CSP ports.
Imagine that node 1 offers a service through port 10 from which it receives **commands** and sends back **response** to its client. Node 2 is a client for node 1, that is, it sends the commands and receives data.

In this scenario, node 2 (client) sends **commands** to node 1 (server) through port 10. But ¿is the server's **response** sent also through port 10?

We believe that it isn't, since "ports 48 to 63 are ephemeral ports used for outgoing connections."

KR

Ángel

yashi, 2024-07-07:
No.

A server usually send a response via the connection taken from `csp_accept()`.  This means that your client gets the response from the connection you created by calling `csp_connect()` on the client side.

CSP has a connection pool and it's initialized at `csp_init()`. `csp_conn_lint()` is called by `csp_init()` and all connections in the poll gets initialized `sport_outgoing` in it.

https://github.com/libcsp/libcsp/blob/3e0e1650872d78cdb85fb7bd0e4540ddce979aff/src/csp_conn.c#L49-L63

`CSP_PORT_MAX_BIND` is a compile-time option you can set for your system, and it defaults to `16`. Thus, your outgoing connections will get 17 and above. I'm not sure where you get the quote:

> ports 48 to 63 are ephemeral ports used for outgoing connections.

Care to share the pointer?

BTW, here are example implementations of server and client as you described:

- server: https://gist.github.com/yashi/15e33944f5b1ec1a1d634b545781b68f
- client: https://gist.github.com/yashi/bd1479a1489d8160a5ee05a9cb174566

yashi, 2024-07-07:
You can initialize ptys using the following command:

```
socat -d -d pty,raw,echo=0,link=/tmp/pty1 pty,raw,echo=0,link=/tmp/pty2
```

AngelPerezM, 2024-07-07:
Hi, @yashi:

Thanks for the quick response, question solved! I got the info about ephemeral ports from Wikipedia: https://en.wikipedia.org/wiki/Cubesat_Space_Protocol#Version_1

KR,
Ángel

closed 2024-07-07 by AngelPerezM