libcsp/libcsp #608: Why my client can't ping to server

poomravint, 2024-07-09:
I am coding two function client() and server() then use pthread to call those functions
Client() will ping to the server but everytime client will get -1 ms 
I don't understand what's point did i do wrong

This is my zip code : 
[csp_re_sen.zip](https://github.com/user-attachments/files/16137674/csp_re_sen.zip)
I set address like this  :
[image]
I think it's my first mistake but I don't know how to set address.
Someone please help me I have been coding this code for 1 month and now It still not working.

yashi, 2024-07-10:
Does the examples on https://github.com/libcsp/libcsp/issues/606#issuecomment-2212402351 help?

poomravint, 2024-07-11:
> Do the examples on [#606 (comment)](https://github.com/libcsp/libcsp/issues/606#issuecomment-2212402351) help?

Thanks a lolt your examples can help me very much.

yashi, 2024-07-11:
Glad to hear that. Repen this or open new one if you have more question. Closing.

closed 2024-07-11 by yashi