libcsp/libcsp #613: Usage of CSP IP stack with the Space Packet Protocol

secureshadow, 2024-07-10:
Hello!

I have a question. If for a software architecture design, in the upper layer (the application layer) I am formating my packages using the CCSDS Space Packet Protocol for using the PUS services, can I use this packet protocol in conjuntion with CSP at a lower layer (in the Network Layer)? Or do I need to implement a middle layer between Application and Network that formats SPP into the IP stack of CSP?

I hope this question is clear.

Thanks in advance!

yashi, 2024-07-10:
If I'm not misunderstanding you, it is certainly possible.

You want to use libcsp as a Data link layer, right?

As in CCSDS 133.0-B-2 Figure 2-1: SPP Context within the CCSDS Protocol Stack

yashi, 2024-07-13:
Reopen it if you have more question. Thanks.

closed 2024-07-13 by yashi