libcsp/libcsp #614: CSP can communicate between 2 process but without interface ?

poomravint, 2024-07-12:
I run csp_server_client.c from your example without interface then it's work .
Then I try to coding csp_server.c and csp_client to communicate between 2 process (no interface) but it's not work

moonlight83340, 2024-07-12:
I really don't think it's aim to be use like that.
I think you should use an interface when client and server aren't in loop-back mode.
I may be wrong so an other answer would be great.

yashi, 2024-07-13:
> I run csp_server_client.c from your example without interface then it's work .

How did you do that?

poomravint, 2024-07-15:
> I really don't think it's aim to be use like that. I think you should use an interface when client and server aren't in loop-back mode. I may be wrong so an other answer would be great.

Thanks

poomravint, 2024-07-15:
> > I run csp_server_client.c from your example without interface then it's work .
> 
> How did you do that? I mean _without (any) interface_.
> 
> I guess you mean you just didn't specify any interface when you ran `csp_server_client`. Even without specifying any, `csp_server_client` creates a special `lo` interface. Since `csp_server_client` is one process, it can communicate internally. Please note that this `lo` interface is not the same interface you see when you run `ip a` on your Linux system.
> 
> Now, if you use `csp_server` and `csp_client`, even though they also create a `lo` interface in their process, libcsp can't communicate between them (I'm assuming Linux). This is because libcsp is not using any kernel feature. The `lo` interface in libcsp is a process-local loopback interface, not the Linux kernel's `lo`.
> 
> On the other hand, if you have two processes, without libcsp, running and communicating using, say, UDP against the ordinary `lo` interface, they can talk to each other. This is because the kernel handles your packet and sends it to the other side. This is not the case for libcsp. It is certainly and technically possible to use Linux's loopback, but nobody has done that. At least for now.

Thanks a lot i got it

closed 2024-07-15 by yashi