libcsp/libcsp #615: Can't build with cmake when enableing python binding on libcsp-2-0 branch

moonlight83340, 2024-07-12:
**Brief**
I'm trying to build with cmake and python binding option libcsp-2-0 branch but I'm unable to do that.

**How to reproduce**
I have clone the libcsp-2-0 branch and I try to build it with python binding.
`cmake -GNinja -B builddir -Denable-python3-bindings=1 -DCSP_HAVE_LIBZMQ=1 -DCSP_USE_RTABLE=1 -DCMAKE_POSITION_INDEPENDENT_CODE=ON && ninja -C builddir`

**Logs**
```
gaetan@gaetan-ThinkPad-T16-Gen-2:~/Downloads/libcsp-2-0/libcsp$ cmake -GNinja -B builddir -Denable-python3-bindings=1 -DCSP_HAVE_LIBZMQ=1 -DCSP_USE_RTABLE=1 -DCMAKE_POSITION_INDEPENDENT_CODE=ON && ninja -C builddir
-- Found Python: /usr/include/python3.10 (found version "3.10.12") found components: Development.Module
-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /home/gaetan/Downloads/libcsp-2-0/libcsp/builddir
ninja: Entering directory `builddir'
[1/2] Building C object CMakeFiles/libcsp_py3.dir/src/bindings/python/pycsp.c.o
/home/gaetan/Downloads/libcsp-2-0/libcsp/src/bindings/python/pycsp.c: In function ‘pycsp_free_csp_socket’:
/home/gaetan/Downloads/libcsp-2-0/libcsp/src/bindings/python/pycsp.c:76:31: warning: initialization of ‘csp_conn_t *’ {aka ‘struct csp_conn_s *’} from incompatible pointer type ‘csp_socket_t *’ {aka ‘struct csp_socket_s *’} [-Wincompatible-pointer-types]
   76 |         csp_conn_t * socket = get_obj_as_socket(obj, true);
      |                               ^~~~~~~~~~~~~~~~~
/home/gaetan/Downloads/libcsp-2-0/libcsp/src/bindings/python/pycsp.c: In function ‘pycsp_socket’:
/home/gaetan/Downloads/libcsp-2-0/libcsp/src/bindings/python/pycsp.c:132:33: warning: implicit declaration of function ‘csp_socket’; did you mean ‘pycsp_socket’? [-Wimplicit-function-declaration]
  132 |         csp_socket_t * socket = csp_socket(opts);
      |                                 ^~~~~~~~~~
      |                                 pycsp_socket
/home/gaetan/Downloads/libcsp-2-0/libcsp/src/bindings/python/pycsp.c:132:33: warning: initialization of ‘csp_socket_t *’ {aka ‘struct csp_socket_s *’} from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
[2/2] Linking C shared module libcsp_py3.cpython-310-x86_64-linux-gnu.so
gaetan@gaetan-ThinkPad-T16-Gen-2:~/Downloads/libcsp-2-0/libcsp$
```

yashi, 2024-07-13:
It must be `-DCSP_ENABLE_PYTHON3_BINDINGS=1` for CMake.

moonlight83340, 2024-07-13:
Looking at the cmakelist : https://github.com/libcsp/libcsp/blob/libcsp-2-0/CMakeLists.txt
it seems not, I have also try with -DCSP_ENABLE_PYTHON3_BINDINGS=1 before.

yashi, 2024-07-13:
Ah, v2.0. For v2.0, python binding is broken I suppose.

@sigmundklaa has fixed it in #411 for `develop` branch.

moonlight83340, 2024-07-13:
Yes, but even with the fix, I'm unable to execute `libcsp.cmp_ident` or `libcsp.transaction`.
I will post more log about my problem later, it could help to solve the python binding not working on v2.

yashi, 2024-07-13:
CMake is not supported in the `libcsp-2-0` branch. Especially for the Python binding. Use Waf or Meson.

**Meson**
```
meson setup builddir -Denable_python3_bindings=true && ninja -C builddir
```

I have to close this as "Won't fix". Sorry.

closed 2024-07-13 by yashi

moonlight83340, 2024-07-14:
I understand that it will be not fixed in the` libcsp-2-0` branch but just to note, build also won't work with meson same error.

yashi, 2024-07-14:
No I didn't say I won't fix for v2.0.  I said CMake isn't supported on v2.0 to build the Python binding. CMake on v2.0 was intended to be Zephyr only. We only gained so many support on `develop` (aka `will-be-v2.1`).

> also won't work with meson same

That's a shame. How about Waf. Or it's just not build system issue?  Let's open a new issue. This issue is about CMake.