libcsp/libcsp #618: Prepare v2.1

yashi, 2024-07-14:
We've tagged v2.0 last April but the branch `libcsp-2-0` has been branched since December 2021, which is more than 2.5 years ago. This means that `develop` branch has already accumeurated 2.5 years of work. So I feel like I should tag v2.1.

I'd like to fix or include following changes before v2.1:

- #617 
- #491 
- #369
- #597 
- #583
- #564

If possible, I'd also want to include

- #586 (waiting for how to test)

If you have any issues or PRs that you would like to be included in v2.1, please let us know.

Thanks.

yashi, 2024-07-17:
I just run [ABI Compliance Checker](https://lvc.github.io/abi-compliance-checker/) on v2.0 and v2.1.  Here is the sumary.

**Problem Summary**

<table border="2" cellspacing="0" cellpadding="6" rules="groups" frame="hsides">

<colgroup>
<col  class="org-left" />

<col  class="org-left" />

<col  class="org-right" />
</colgroup>
<tbody>
<tr>
<td class="org-left">&#xa0;</td>
<td class="org-left">Severity</td>
<td class="org-right">Count</td>
</tr>

<tr>
<td class="org-left">Added Symbols</td>
<td class="org-left">-</td>
<td class="org-right">16</td>
</tr>

<tr>
<td class="org-left">Removed Symbols</td>
<td class="org-left">High</td>
<td class="org-right">25</td>
</tr>

<tr>
<td class="org-left">Problems with Data Types</td>
<td class="org-left">High</td>
<td class="org-right">0</td>
</tr>

<tr>
<td class="org-left">&#xa0;</td>
<td class="org-left">Medium</td>
<td class="org-right">5</td>
</tr>

<tr>
<td class="org-left">&#xa0;</td>
<td class="org-left">Low</td>
<td class="org-right">16</td>
</tr>

<tr>
<td class="org-left">Problems with Symbols</td>
<td class="org-left">High</td>
<td class="org-right">12</td>
</tr>

<tr>
<td class="org-left">&#xa0;</td>
<td class="org-left">Medium</td>
<td class="org-right">11</td>
</tr>

<tr>
<td class="org-left">&#xa0;</td>
<td class="org-left">Low</td>
<td class="org-right">12</td>
</tr>

<tr>
<td class="org-left">Problems with Constants</td>
<td class="org-left">Low</td>
<td class="org-right">0</td>
</tr>
</tbody>
</table>

Or you can read this HTML.
[compat_report.html.gz](https://github.com/user-attachments/files/16263077/compat_report.html.gz)

It'd be nice if sone one integrate this in GitHub Actions.

dimitrovand, 2024-07-18:
@yashi The following PR can be included as well https://github.com/libcsp/libcsp/pull/503

johandc, 2024-10-02:
The semantic versioning needs to be interpreted differently. Here the Major version number relates to the Header definition of the packets. Not the ABI of the library.

johandc, 2024-10-02:
Due to the recently found CRC32 bug, we need to extend the period of which packets are transmitted without header CRC. Because clients up untill two months ago would not support decoding it correctly. One suggestion is to tag 2.1 now, with the fixed CRC32 verify and wait till 2.2 with including header as default.

For "fast movers" we should hoever add support for setting the new format, as a runtime init parameter. Because we would like to transition quite soon.

yashi, 2024-12-02:
I've removed #617 from v2.1.

yashi, 2025-07-18:
@JulianTiller, sorry for the delay. I’ve been busy since last December and am only now recovering. We have a lot of issues to address at the moment, and I’m working on triaging them. Let’s see how it goes.

yashi, 2025-07-22:
OK. Once we make `csp_buffer_get_always()` internal function, I'll cut v2.1. We can fix any bugs in v2.1 branch if needed. If you want any fix before the release, please send me a PR.

eg321, 2025-10-01:
Guys, hi!
Can you share plans about releasing v2.1 and 2.2 please?

As we can see, there a lot of changes in `develop` branch that can be useful, but your release process is not obvious.
Thanks.