libcsp/libcsp #619: python: Can't build python binding on libcsp-2-0 branch

moonlight83340, 2024-07-14:
So far I have try Cmake and meson to build it.
`cmake -GNinja -B builddir -Denable-python3-bindings=1 -DCSP_HAVE_LIBZMQ=1 -DCSP_USE_RTABLE=1 -DCMAKE_POSITION_INDEPENDENT_CODE=ON && ninja -C builddir`

**Brief**
I'm trying to build `libcsp-2-0` branch and python binding option libcsp-2-0 branch but I'm unable to do that.

**How to reproduce**
I have clone the libcsp-2-0 branch and I try to build it with python binding.
`cmake -GNinja -B builddir -Denable-python3-bindings=1 -DCSP_HAVE_LIBZMQ=1 -DCSP_USE_RTABLE=1 -DCMAKE_POSITION_INDEPENDENT_CODE=ON && ninja -C builddir`
or with meson
`meson setup builddir -Denable_python3_bindings=true && ninja -C builddir`

**OS**
Ubuntu 22

**Logs**
```
gaetan@gaetan-ThinkPad-T16-Gen-2:~/Downloads/libcsp-2-0/libcsp$ cmake -GNinja -B builddir -Denable-python3-bindings=1 -DCSP_HAVE_LIBZMQ=1 -DCSP_USE_RTABLE=1 -DCMAKE_POSITION_INDEPENDENT_CODE=ON && ninja -C builddir
-- Found Python: /usr/include/python3.10 (found version "3.10.12") found components: Development.Module
-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /home/gaetan/Downloads/libcsp-2-0/libcsp/builddir
ninja: Entering directory `builddir'
[1/2] Building C object CMakeFiles/libcsp_py3.dir/src/bindings/python/pycsp.c.o
/home/gaetan/Downloads/libcsp-2-0/libcsp/src/bindings/python/pycsp.c: In function ‘pycsp_free_csp_socket’:
/home/gaetan/Downloads/libcsp-2-0/libcsp/src/bindings/python/pycsp.c:76:31: warning: initialization of ‘csp_conn_t *’ {aka ‘struct csp_conn_s *’} from incompatible pointer type ‘csp_socket_t *’ {aka ‘struct csp_socket_s *’} [-Wincompatible-pointer-types]
   76 |         csp_conn_t * socket = get_obj_as_socket(obj, true);
      |                               ^~~~~~~~~~~~~~~~~
/home/gaetan/Downloads/libcsp-2-0/libcsp/src/bindings/python/pycsp.c: In function ‘pycsp_socket’:
/home/gaetan/Downloads/libcsp-2-0/libcsp/src/bindings/python/pycsp.c:132:33: warning: implicit declaration of function ‘csp_socket’; did you mean ‘pycsp_socket’? [-Wimplicit-function-declaration]
  132 |         csp_socket_t * socket = csp_socket(opts);
      |                                 ^~~~~~~~~~
      |                                 pycsp_socket
/home/gaetan/Downloads/libcsp-2-0/libcsp/src/bindings/python/pycsp.c:132:33: warning: initialization of ‘csp_socket_t *’ {aka ‘struct csp_socket_s *’} from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
[2/2] Linking C shared module libcsp_py3.cpython-310-x86_64-linux-gnu.so
gaetan@gaetan-ThinkPad-T16-Gen-2:~/Downloads/libcsp-2-0/libcsp$
```

Same error happened with cmake and meson.

I will try to give a try with Waf but I can't at the time. (Maybe next week)

yashi, 2024-07-14:
`git bisect` told me that the issue was fixed by cf8650c5f2d28f3e471e7e8cefd7346110d9a10a on `develop`.

moonlight83340, 2024-07-16:
> `git bisect` told me that the issue was fixed by [cf8650c](https://github.com/libcsp/libcsp/commit/cf8650c5f2d28f3e471e7e8cefd7346110d9a10a) on `develop`.

Adding this commit, helped to build. It build.
But I'm unable to use the python binding, csp_cmp_ident / ping / transaction failed everytime.

closed 2024-08-18 by moonlight83340

yashi, 2024-08-18:
@moonlight83340 Did you open another issue? if so, would you mind mention it here?

moonlight83340, 2024-08-19:
I open a new issue https://github.com/libcsp/libcsp/issues/664 which is related with this issue because I found that even if libcsp-2-0 branch can build,there is an error with cmp_ident.