libcsp/libcsp #620: public API thread safety

dimitrovand, 2024-07-22:
The following public API methods appear to be non-thread-safe. Note that there may be others I have overlooked:
```
csp_rdp_set_opt()
csp_rdp_get_opt()
csp_cmp_set_memcpy()
csp_bridge_set_interfaces()
csp_bind_callback()
csp_bind()
csp_socket_close()
all functions from csp_rtable.h
most functions from csp_iflist.h 
```
There is currently no information indicating whether or not these functions are intended to be thread-safe. As it stands, users must individually verify the implementation of each function before using them in a multi-threaded environment. This adds an extra layer of complexity and potential risk for developers working with these APIs.

yashi, 2024-08-06:
We really need this verified.

johandc, 2024-10-02:
> csp_rdp_set_opt() 
> csp_rdp_get_opt()

Verified these are an issue. We could make a version of the csp_connect, which passes rdp parameters. Instead of global options. Using these could cause a race condition.

> csp_cmp_set_memcpy()
> csp_bridge_set_interfaces()
> all functions from csp_rtable.h
> most functions from csp_iflist.h 
> csp_bind_callback()

All of the above is not expected to be thread safe. Because they should only be used during initialization of the CSP stack.

> csp_bind()
> csp_socket_close()

This does not need to be thread safe, because it is never allowed to bind two sockets to the same port.
Similarly, when a socket is bound to a port and closed, there is no conflicts with other tasks on the cso_socket_close call.

dimitrovand, 2024-10-07:
Thanks for the clarification!

closed 2024-10-07 by dimitrovand