libcsp/libcsp #621: CSP with CAN interface can't compile

poomravint, 2024-07-26:
I have include can_socketcan.h and csp_if_can.h already
[image]
But when I try to compile I got this : 
[image]

And I don't understand what parameters that I have to use with csp_can_socketcan_open_and_add_interface()
[image]

dimitrovand, 2024-07-26:
Hello @poomravint.
From your screenshots, it appears that the linker can't find the implementation of `csp_can_socketcan_open_and_add_interface`. To resolve this issue, you likely need to have `libsocketcan` installed on your host machine when compiling `libcsp`.
To verify whether `libsocketcan` is present, simply recompile `libcsp` and check the output log of the compilation.

About `csp_can_socketcan_open_and_add_interface` input parameters:
1. name of the can device. This is the netdev name, as ifconfig -a shows in your system. usually it contains prefix "can" and the numer of the can line. e.g. "can0" (you can check the docs here https://lalten.github.io/libsocketcan/Documentation/html/group__extern.html#ga0df08d0fdd25560a7c66923e48acda55)
2. string representing the name of your CAN interface (e.g. "MY_CSP_CAN_DEV")
3. CSP node address of your interface (i don't think you can set it to 0)
4. desired CAN bus speed/bitrate
5. When true, the interface will receive all CAN frames, regardless of their destination address. When false, the interface will only receive CAN frames that match its own address.
6. will contain a reference to your newly created interface (it is an optional parameter)

poomravint, 2024-07-30:
> Hello @poomravint. From your screenshots, it appears that the linker can't find the implementation of `csp_can_socketcan_open_and_add_interface`. To resolve this issue, you likely need to have [...]

How to install libsocketcan on RasPI . I got install libsocketcan on my RasPI but I still can't compile
[image]
Or maybe I got install wrong lib this is my lib that I installed on my PI
[image]

dimitrovand, 2024-07-30:
You should be fine with running `sudo apt install libsocketcan-dev.` Did you recompile `libcsp`? If so, what was the output when you executed `cmake -G Ninja -B builddir`?

poomravint, 2024-07-30:
> You should be fine with running `sudo apt install libsocketcan-dev.` Did you recompile `libcsp`? If so, what was the output when you executed `cmake -G Ninja -B builddir`?

What is mean recompile libcsp ?
This is my output when I executed `cmake -G Ninja -B builddir`
[image]

poomravint, 2024-07-30:
> > You should be fine with running `sudo apt install libsocketcan-dev.` Did you recompile `libcsp`? If so, what was the output when you executed `cmake -G Ninja -B builddir`?
> 
> What is mean recompile libcsp ? If you mean try to compile again after `run sudo apt install libsocketcan-dev` I got this :
[image]
This is my output when I executed `cmake -G Ninja -B builddir` ![image](https://private-user-images.githubusercontent.com/101710514/353343717-b4d922c0-c3cd-4926-b08a-48f9ed8333de.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIzMjE4NTgsIm5iZiI6MTcyMjMyMTU1OCwicGF0aCI6Ii8xMDE3MTA1MTQvMzUzMzQzNzE3LWI0ZDkyMmMwLWMzY2QtNDkyNi1iMDhhLTQ4ZjllZDgzMzNkZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDczMFQwNjM5MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZmQ0ZGM1YTMwZjUwZDhhNWU2ZjViZTkxNmUwZmMwYmVjMDdiOGIwYTFjYmVlNjAxNjlmM2QyMGU3MDgzYjVlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.SC19CxUxho_7D0Wh7OhBrqRT2VafT8LQazxMDCBmG4A) What should I do next ?

dimitrovand, 2024-07-30:
Ok, as you can see, it says `Found /usr/.../libsocketcan.so 0.0.12,` which means CMake has detected libsocketcan. You can now follow the instructions at https://libcsp.github.io/libcsp/INSTALL.html to build and install libcsp. Finally, recompile your application, and the undefined reference issue should be resolved.

poomravint, 2024-07-30:
> Ok, as you can see, it says `Found /usr/.../libsocketcan.so 0.0.12,` which means CMake has detected libsocketcan. You can now follow the instructions at https://libcsp.github.io/libcsp/INSTALL.html [...]

I have build and intsalled libcsp for one time before I run `sudo apt install libsocketcan-dev` I have to delete libcsp before build and install libcsp again or Can I build and install libcsp ?

dimitrovand, 2024-07-30:
Either option should work just fine.

poomravint, 2024-07-30:
> Either option should work just fine.

Thanks but I have some question How to uninstall libcsp , I will build and install again

closed 2024-07-31 by poomravint