libcsp/libcsp #623: Unable to run python exmaples because libcsp_py3 has an invalid symbol

Samalmeida1028, 2024-07-31:
when running the python examples I get this error:

import libcsp_py3 as csp
ImportError: build/libcsp.so: undefined symbol: yaml_parser_set_input_file

yashi, 2024-07-31:
Thanks.

This works. (Waf)
```
❯ rm -rf build && ./examples/buildall.py && PYTHONPATH=build python3 -c 'import libcsp_py3 as csp'
```

But, neither Meson nor CMake work:

**Meson**
```
❯ rm -rf build && meson setup -Denable_python3_bindings=true build && ninja -C build && PYTHONPATH=build python3 -c 'import libcsp_py3 as csp'
The Meson build system
Version: 1.5.0
Source dir: /home/yashi/src/libcsp
Build dir: /home/yashi/src/libcsp/build
Build type: native build
Project name: csp
Project version: 2.1
C compiler for the host machine: sccache cc (gcc 14.1.0 "cc (Debian 14.1.0-5) 14.1.0")
C linker for the host machine: cc ld.bfd 2.42.90.20240720
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: YES (/usr/bin/pkg-config) 1.8.1
Run-time dependency yaml-0.1 found: YES 0.2.5
Library c found: YES
Run-time dependency threads found: YES
Run-time dependency libsocketcan found: YES 0.0.12
Run-time dependency libzmq found: YES 4.3.5
Has header "sys/socket.h" : YES 
Has header "arpa/inet.h" : YES 
Configuring autoconfig.h using configuration
Program python3 found: YES (/usr/bin/python3)
Run-time dependency python3 found: YES 3.12
Run-time dependency python found: YES 3.12
meson.build:103: WARNING: Project targets '>=0.53.2' but uses feature introduced in '0.63.0': python_installation.extension_module with implicit dependency on python. use python_installation.dependency()
Build targets in project: 7
WARNING: Project specifies a minimum meson_version '>=0.53.2' but uses features which were added in newer versions:
 * 0.63.0: {'python_installation.extension_module with implicit dependency on python'}

csp 2.1

  User defined options
    enable_python3_bindings: true

Found ninja-1.12.1 at /usr/bin/ninja
ninja: Entering directory `build'
[47/47] Linking target libcsp_py3.cpython-312-x86_64-linux-gnu.so
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: /home/yashi/src/libcsp/build/libcsp_py3.cpython-312-x86_64-linux-gnu.so: undefined symbol: csp_rtable_clear
```

**CMake**
```
❯ rm -rf build && cmake -B build -DCSP_ENABLE_PYTHON3_BINDINGS=1 && ninja -C build && PYTHONPATH=build python3 -c 'import libcsp_py3 as csp'
-- The C compiler identification is GNU 14.1.0
-- The CXX compiler identification is GNU 14.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/lib/ccache/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib/ccache/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to MinSizeRel
-- Looking for include file sys/socket.h
-- Looking for include file sys/socket.h - found
-- Looking for include file arpa/inet.h
-- Looking for include file arpa/inet.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
-- Checking for one of the modules 'yaml-0.1'
-- Found /usr/lib/x86_64-linux-gnu/libyaml.so 0.2.5
-- Checking for one of the modules 'libzmq'
-- Found /usr/lib/x86_64-linux-gnu/libzmq.so 4.3.5
-- Checking for one of the modules 'libsocketcan'
-- Found /usr/lib/x86_64-linux-gnu/libsocketcan.so 0.0.12
-- Found Python3: /usr/include/python3.12 (found version "3.12.4") found components: Development.Module
-- Configuring done (0.5s)
-- Generating done (0.0s)
-- Build files have been written to: /home/yashi/src/libcsp/build
ninja: Entering directory `build'
[43/43] Linking C shared module libcsp_py3.cpython-312-x86_64-linux-gnu.so
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: /home/yashi/src/libcsp/build/libcsp_py3.cpython-312-x86_64-linux-gnu.so: undefined symbol: csp_rtable_clear
```

yashi, 2024-08-06:
OK. These works

**Meson**
```
❯ rm -rf build builddir && meson setup -Denable_python3_bindings=true -Duse_rtable=true build && ninja -C build && PYTHONPATH=build python3 -c 'import sys; print(sys.version); import libcsp_py3 as csp'
```

**CMake**
```
❯ rm -rf build && cmake -B build -DCSP_ENABLE_PYTHON3_BINDINGS=1 -DCSP_USE_RTABLE=1 && ninja -C build && PYTHONPATH=build python3 -c 'import libcsp_py3 as csp'
```

**Discussion**
We can make `ifdef` every feature in the python bindings. But does it worth it? How about enabling RTABLE in libcsp by default? These days, unused functions are remove at link time and won't be included in the final image.

I think it's a good idea to make sure the default build just works, though.

PR #626 (2024-08-07): python: Conditional RTABLE-related functions
RTABLE is disabled by default, but the Python bindings unconditionally call these functions. Added ifdef statements around RTABLE functions.

This fixes #623.
closed 2024-08-07 by yashi