libcsp/libcsp #636: Python to PyPI

FelixEL, 2024-08-11:
Will pythonic version of this library be published on pyPI?

yashi, 2024-08-12:
I'm not really a python guy.  A PR is always welcome :smile:

parker-research, 2024-08-17:
+1 for this. As long as there's no disagreement with it, I'd be happy to help prep this! Is there an account management system to do this? We'd need a `libcsp` PyPI.org account, of course.

Is the Python code versioned alongside the main `libcsp`, or should it get its own version numbers?

johandc, 2024-10-02:
I think we should aim for a Native Python implementation of CSP instead.

Lots of what is going on inside CSP, the router, the many network interfaces, the low level OS stuff, is really gone when inside python context. Most likely the main interface used with  python is going to be ZMQ which is already native python as well.

DeflateAwning, 2024-10-03:
I think this is good enough for now. Really targeting ease of use for ground stations would be huge

kivkiv12345, 2024-10-21:
A more immediate alternative could be to create a [pyproject.toml](https://github.com/spaceinventor/PyCSH/blob/master/pyproject.toml) file, allowing the repository itself to be pip3 installed.

I have tried this myself with the Space Inventor repository [PyCSH](https://github.com/spaceinventor/PyCSH), although here we mostly use an [./install](https://github.com/spaceinventor/PyCSH/blob/master/install) script to pip3 install a local build. But a GitHub URL can also be used.

DeflateAwning, 2024-10-21:
That is indeed the first step!

kivkiv12345, 2024-10-23:
I've made a PR for the pyproject.toml part:
https://github.com/libcsp/libcsp/pull/714
I took a creative liberty by bundling libcsp.so into the pip wheel,
but I think this will be greatly appreciated when users can install directly from PyPI without needing to install libcsp on their system first.

parker-research, 2024-10-23:
Would be awesome to get a backport to v1.6 for this, and have that lineage published to PyPI!

kivkiv12345, 2024-10-24:
I think you can configure the newest commit to CSP 1.6, right?

parker-research, 2024-10-24:
I'm just not too sure if all the other changes in that PR make sense to include in the backport; can you advise? Are any of them breaking changes?

kivkiv12345, 2024-10-24:
* I don't know whether the bindings API has seen breaking changes since 1.6 (outside of this PR), where the .pyi file would need to be backported.
* I will probably have to admit to one breaking change. I've left a comment on the PR mentioning that marking pycsp_buffer_get() with METH_NOARGS is a breaking change, as some users may still call it with the unused size argument.
So I will probably change the function to handle that one argument specifically, letting other invalid calls raise exceptions.
* I don't intend to make any breaking changes by bundling libcsp.so, as symbols from either the existing namespace or system libraries should take priority when importing/linking. However I am still testing this, in the meantime I would love some discussion as to what others think of this approach.