libcsp/libcsp #650: RFC: Enable `-pedantic`

yashi, 2024-08-17:
## Description

@parker-research has found that we have a few minor errors in our code base which can be found with `-pedantic`.

- #645 
- #646 

We also have fixed a few others:

- #363
- https://github.com/libcsp/libcsp/commit/ed71a2a80f60b8366b849eb8353fe14e007d8710

Should we enable `-pedantic` by default?

## Steps to Reproduce

1. `cmake -B builddir -DCMAKE_C_FLAGS="-pedantic"`
2. `ninja -C builddir`

## Expected behavior

No error even with `-pedantic`

## Actual Behavior:

```
ninja: Entering directory `builddir'
[7/41] Building C object CMakeFiles/csp.dir/src/csp_conn.c.o
/home/yashi/src/libcsp/src/csp_conn.c: In function ‘csp_conn_print_table_str’:
/home/yashi/src/libcsp/src/csp_conn.c:390:52: warning: format ‘%p’ expects argument of type ‘void *’, but argument 5 has type ‘csp_conn_t *’ {aka ‘struct csp_conn_s *’} [-Wformat=]
  390 |                 snprintf(buf, sizeof(buf), "[%02u %p] S:%u, %u -> %u, %u -> %u (%u)\n",
      |                                                   ~^
      |                                                    |
      |                                                    void *
  391 |                                  i, conn, conn->state, conn->idin.src, conn->idin.dst,
      |                                     ~~~~            
      |                                     |
      |                                     csp_conn_t * {aka struct csp_conn_s *}
[25/41] Building C object CMakeFiles/csp.dir/src/arch/posix/csp_queue.c.o
/home/yashi/src/libcsp/src/arch/posix/csp_queue.c: In function ‘csp_queue_empty’:
/home/yashi/src/libcsp/src/arch/posix/csp_queue.c:46:16: warning: ISO C forbids ‘return’ with expression, in function returning void [-Wpedantic]
   46 |         return pthread_queue_empty(handle);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/yashi/src/libcsp/src/arch/posix/csp_queue.c:45:6: note: declared here
   45 | void csp_queue_empty(csp_queue_handle_t handle) {
      |      ^~~~~~~~~~~~~~~
[41/41] Linking C shared library libcsp.so
```

## Environment Details:

Provide information about your environment, including:

- Build OS: Debian Sid
- Target OS: Linux
- libcsp commit hash: `develop (v1.6-1128-g4c4fd14c)
- CMake version: 3.30.2
- Ninja version: 1.12.1

## Logs or console output from the very beginning to the end

```
❯ rm -rf builddir && cmake -B builddir -DCMAKE_C_FLAGS="-pedantic" && ninja -C builddir
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/lib/ccache/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib/ccache/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to MinSizeRel
-- Looking for include file sys/socket.h
-- Looking for include file sys/socket.h - found
-- Looking for include file arpa/inet.h
-- Looking for include file arpa/inet.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
-- Checking for one of the modules 'yaml-0.1'
-- Found /usr/lib/x86_64-linux-gnu/libyaml.so 0.2.5
-- Checking for one of the modules 'libzmq'
-- Found /usr/lib/x86_64-linux-gnu/libzmq.so 4.3.5
-- Checking for one of the modules 'libsocketcan'
-- Found /usr/lib/x86_64-linux-gnu/libsocketcan.so 0.0.12
-- Configuring done (0.5s)
-- Generating done (0.0s)
-- Build files have been written to: /home/yashi/src/libcsp/builddir
ninja: Entering directory `builddir'
[7/41] Building C object CMakeFiles/csp.dir/src/csp_conn.c.o
/home/yashi/src/libcsp/src/csp_conn.c: In function ‘csp_conn_print_table_str’:
/home/yashi/src/libcsp/src/csp_conn.c:390:52: warning: format ‘%p’ expects argument of type ‘void *’, but argument 5 has type ‘csp_conn_t *’ {aka ‘struct csp_conn_s *’} [-Wformat=]
  390 |                 snprintf(buf, sizeof(buf), "[%02u %p] S:%u, %u -> %u, %u -> %u (%u)\n",
      |                                                   ~^
      |                                                    |
      |                                                    void *
  391 |                                  i, conn, conn->state, conn->idin.src, conn->idin.dst,
      |                                     ~~~~            
      |                                     |
      |                                     csp_conn_t * {aka struct csp_conn_s *}
[25/41] Building C object CMakeFiles/csp.dir/src/arch/posix/csp_queue.c.o
/home/yashi/src/libcsp/src/arch/posix/csp_queue.c: In function ‘csp_queue_empty’:
/home/yashi/src/libcsp/src/arch/posix/csp_queue.c:46:16: warning: ISO C forbids ‘return’ with expression, in function returning void [-Wpedantic]
   46 |         return pthread_queue_empty(handle);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/yashi/src/libcsp/src/arch/posix/csp_queue.c:45:6: note: declared here
   45 | void csp_queue_empty(csp_queue_handle_t handle) {
      |      ^~~~~~~~~~~~~~~
[41/41] Linking C shared library libcsp.so
```

## Checklist

- [x] I've searched in "Issues" for similar issues, both open and closed.
- [x] I've used GitHub Flavored Markdown (GFM) to format my text.
- [x] I've avoided using screenshots for console output or logs
- [x] I've included all relevant details and steps to reproduce the issue.
- [x] I've provided information about my environment.
- [x] I've attached necessary logs or code snippet for minimal reproducible example.

parker-research, 2024-08-17:
I personally support enabling `-Wpedantic`. As a project intended for space systems (which generally have very high reliablity requirements), having the extra checks is very important/valuable. I'd expect that many softwares using this framework aim to have the `-Wpedantic` flag enabled on the entire build (like our [CTS-SAT-1 Onboard Computer Firmware](https://github.com/CalgaryToSpace/CTS-SAT-1-OBC-Firmware)), so by extension, this project should support it.

yashi, 2024-08-18:
I personally haven't had good experiences with `-pedantic`. To me, it mostly generates non-critical warnings and complains about the use of GCC extensions. While it's true that we aren't using any GCC extensions at the moment, so enabling it might seem like a good idea, I'm still uncertain.

For example:

- Git doesn't enable `-pedantic` in ordinal build.  It's [only for developers](https://github.com/git/git/blob/b9849e4f7631d80f146d159bf7b60263b3205632/config.mak.dev#L12).
- NASA [cFS](https://github.com/nasa/cFS)/[cFE](https://github.com/nasa/cFE) also doesn't use `-pedantic` on all modules.
- NASA [F'](https://github.com/nasa/fprime) doesn't use `-pedantic` with GCC but only with Clang.

Is anyone using compilers other than GCC or Clang who has encountered real, practical issues because we don't comply with `-pedantic`?

parker-research, 2024-08-19:
> While it's true that we aren't using any GCC extensions at the moment, so enabling it might seem like a good idea, I'm still uncertain.

Writing code that conforms to the base C spec makes it more likely to work (and easier to port) across compilers, microcontrollers, etc. The cost of enabling it is very cheap; nearly all the issues are fixed. The cost of each individual team enabling it by choice and having to fix the issues themselves is comparatively more expensive.

NASA uses `-pedantic` on some modules, meaning that it has value, at least in places where there's no/limitted friction to enable it (presumably in new modules?).

Not certain it was caught with `pedantic`, but the bug fixed in #647 (which existed for many years unnoticed) is finally fixed. 

Space stuff is really expensive, and C has a lot of quirks. Anything we can do improve the safety of systems built with C for space can and should be done.

PR #718 (2024-11-01): Enable -Wpedantic
I believe that for libraries like libcsp, it's beneficial to avoid GNU extensions and other non-standard features. The `-Wpedantic` flag helps enforce this by warning when such extensions are used.

This PR has fixes found with `-Wpendatic` and updates for all build systems.

This fixes #650.
closed 2024-11-01 by yashi