libcsp/libcsp #664: python: python binding example failed to connect on libcsp-2-0 branch

moonlight83340, 2024-08-19:
## Description

I'm currently having an error when trying to test python binding on libcsp-2-0 branch.
csp_cmp_ident failed with error `-3` .

## Steps to Reproduce

1. Build libcsp and build libcsp python binding
2. Then run `build/examples/zmqproxy `
3. Then `PYTHONPATH=builddir python3 examples/python_bindings_example_server.py` 
4. Then `PYTHONPATH=builddir python3 examples/python_bindings_example_client.py -z localhost -s 27 -a 2`

## Expected behavior

I expected the python binding work with no error.

## Actual Behavior:

csp_cmp_ident failed with error `-3` .
See log.

## Environment Details:

Provide information about your environment, including:

- Build OS:  Ubuntu 22.04
- Target OS: Ubuntu 22.04
- libcsp commit hash: 87f205b30d9a05dcf9afb0bfd50975a77ce3761a
- Meson version: 0.53.2-2ubuntu2
- Ninja version: 1.10.0-1build1

## Logs or console output from the very beginning to the end

```
   File "/home/runner/work/libcsp/libcsp/examples/python_bindings_example_client.py", line 79, in <module>
    print("CMP ident:", libcsp.cmp_ident(options.server_address))
csp.Error: csp_cmp_ident(), result/error: -3
```

## Additional Context:

I'm trying to make the python workflow works on libcsp-2-0 branch, and I'm getting an error on csp_cmp_ident().
I think it may be a connection problem between server and client.

Server address is `27` by default.

  `build/examples/zmqproxy `
  `PYTHONPATH=builddir python3 examples/python_bindings_example_server.py` 
  `PYTHONPATH=builddir python3 examples/python_bindings_example_client.py -z localhost -s 27 -a 2`
  
 
  See my modification : https://github.com/moonlight83340/libcsp/commits/libcsp-2-0-python-binding-test-clean/
  
  
## Checklist

- [ x] I've searched in "Issues" for similar issues, both open and closed.
- [x ] I've used GitHub Flavored Markdown (GFM) to format my text.
- [x ] I've avoided using screenshots for console output or logs
- [x ] I've included all relevant details and steps to reproduce the issue.
- [x ] I've provided information about my environment.
- [x ] I've attached necessary logs or code snippet for minimal reproducible example.

yashi, 2024-08-19:
Where is the log? It says "Logs or console output from the **_very beginning_** to the end" but you don't have it.

moonlight83340, 2024-08-22:
Fix by https://github.com/libcsp/libcsp/pull/665

closed 2024-08-22 by moonlight83340