libcsp/libcsp #695: libcsp drops received packet via CAN wrongly

takumiando, 2024-10-02:
## Description

This is related to https://github.com/libcsp/libcsp/issues/531.

```last_used``` is set as current(last used) time, so it shouldn't be set as ```0```.    
But it is sharing storage area with ```timestamp_rx```, and it is set as ```0```.

So some received packet via CAN will be timed out  in ```csp_can_pbuf_cleanup()``` wrongly.

## Steps to Reproduce

I tried this modified ```csp_client``` and ```csp_server``` examples with CSP version 1.
https://github.com/takumiando/libcsp/tree/last_used_example

With CSP version 2, this problem doesn't occur(why?).

### Server

```shell
$ ./build/examples/csp_server -c vcan0 -a 1
```

### Client 1

```
$ ./build/examples/csp_client -c vcan0 -a 2 -C 1
```
### Client 2

```shell
$ ./build/examples/csp_client -c vcan0 -a 3 -C 1
```

### Result

The server drops some packets because ```last_used``` is ```0```.

```
2:ABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQ
RSTUVWXYZABCDEFGHIJKLMNOPQRST                                                                                                                                                                                                      
pbuf is cleaned up: last_used: 0, now: 174380744        
3:ABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQ
RSTUVWXYZABCDEFGHIJKLMNOPQRST                           
3:ABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQ
RSTUVWXYZABCDEFGHIJKLMNOPQRST                           
3:ABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQ
RSTUVWXYZABCDEFGHIJKLMNOPQRST                           
3:ABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQ
RSTUVWXYZABCDEFGHIJKLMNOPQRST                           
pbuf is cleaned up: last_used: 0, now: 174380765                                                                                                                                                                                   
pbuf is cleaned up: last_used: 0, now: 174380775        
pbuf is cleaned up: last_used: 0, now: 174380807
...
```

- Why is ```last_used``` sharing storage area with ```timestamp_rx```?
- Why does not this problem occur with CSP verson 1?

takumiando, 2024-10-09:
>Why does not this problem occur with CSP version 1?

The timing of the execution of ```csp_id_strip()``` differs between CSP V1 and V2.

https://github.com/libcsp/libcsp/blob/79b2294e8b68c802902005bd400d62460a7ad7da/src/interfaces/csp_if_can.c#L94
https://github.com/libcsp/libcsp/blob/79b2294e8b68c802902005bd400d62460a7ad7da/src/interfaces/csp_if_can.c#L358

In CSP V1, ```csp_id_strip()``` may be executed before receiving the continuation of a CAN packet.  
Since ```csp_id_strip()``` sets ```timestamp_rx``` to 0, ```last_used``` also becomes 0.  
So ```csp_can_pbuf_cleanup()``` timed out the packet.

It is still unclear whether the timing of executing ```csp_id_strip()``` in CSP V1 is wrong.  
However, unless there is a strong reason to union ```timestamp_rx``` and ```last_used```, I think they should be separated into different variables.

yashi, 2024-11-01:
This should have been fixed by #717. I'm closing it. Please reopen if not.