libcsp/libcsp #698: Requirements and Coverage

perigoso, 2024-10-03:
Hello,

I'm investigating the possibility of integrating this library in a professional setting, but we have some requirements that it is not immediately obvious this meets:

- Are there any formal requirements defined?
- Is there any coverage testing/reporting?
- Are there any tests implemented beyond  the workflows in GitHub actions?

Thanks in advance :smiley:

yashi, 2024-10-07:
Hi @perigoso,

> * Are there any formal requirements defined?

AFAIK, no.

> * Is there any coverage testing/reporting?

no, not in the public.

> * Are there any tests implemented beyond  the workflows in GitHub actions?

no. I've started https://github.com/spacecubics/libcsp-zephyr but progress is slow.

yashi, 2024-10-16:
Please reopen if you need more info. closing.

closed 2024-10-16 by yashi

perigoso, 2024-10-16:
Thank you for the info