libcsp/libcsp #703: Where should I add the csp_service_handler?

hernaangq, 2024-10-11:
Hello,

I am currently running a CSP network with 2 successful servers: One binded to port 10 and the other bind to port 11.

I am wondering **what should I do in order to run the default CSP service handler to handle pings, buffer use, etc.**

I suppose that if I make a third server and bind the CSP_ANY, packets from all ports (including the ones to port 10 and port 11) will get there and I don´t want that because that won´t be efficient.

So what should I do?
1. Use one of the current servers, i.e. the one in port 10 and change to ANY so CSP service handler lies there.
2. Or make this third server with CSP_ANY in a possibly inefficient way.

Thanks.
Hernan

yashi, 2024-10-16:
I'm not sure I fully understand your situation.

Are you running multiple server processes on Linux, each with different addresses and ports? What exactly do you mean by "server"? Please describe your system in more detail.

Based on the information provided, the simple answer to your question would be: "Wherever it works for you."

hernaangq, 2024-10-22:
I solved the issue with a second CSP server, for anyone's interest.

Packages that fit in one port won´t fit again in CSP_ANY. 

Thanks for your response.

closed 2024-10-22 by hernaangq