libcsp/libcsp #708: rx_count is broken because of union

takumiando, 2024-10-15:
## Description

This is related to https://github.com/libcsp/libcsp/issues/531 and https://github.com/libcsp/libcsp/issues/695.

```rx_count``` has been broken with the commit https://github.com/libcsp/libcsp/commit/b74b25ec3757a4c8c39a54d25f4d9d8de731f6e9.  
So libcsp V1 cannot receive data over 2 bytes correctly.

Unless there is a strong reason, we shouldn't use union in ```csp_packet_s```.

takumiando, 2024-10-15:
This bug is similar to https://github.com/libcsp/libcsp/issues/695.

The timing of the execution of ```csp_id_strip()```(this means ```rx_timestamp = 0```) differs between CSP V1 and V2.  
So ```cfpid``` will not be broken in V2.

If we can freely modify either union, there’s no reason to use union.

yashi, 2024-10-15:
Please provide steps to reproduce the bug.

takumiando, 2024-10-15:
@yashi 

### Apply this patch to use CSP v1

```diff
diff --git a/examples/csp_client.c b/examples/csp_client.c
index 37625a3..b59a804 100644
--- a/examples/csp_client.c
+++ b/examples/csp_client.c
@@ -139,6 +139,8 @@ int main(int argc, char * argv[]) {
        int ret = EXIT_SUCCESS;
     int opt;
 
+       csp_conf.version = 1;
+
        while ((opt = getopt_long(argc, argv, OPTION_c OPTION_z OPTION_R "k:a:C:tT:h", long_options, NULL)) != -1) {
         switch (opt) {
             case 'c':
diff --git a/examples/csp_server.c b/examples/csp_server.c
index e124c28..a42eb4a 100644
--- a/examples/csp_server.c
+++ b/examples/csp_server.c
@@ -184,6 +184,8 @@ int main(int argc, char * argv[]) {
        csp_iface_t * default_iface;
     int opt;
 
+       csp_conf.version = 1;
+
        while ((opt = getopt_long(argc, argv, OPTION_c OPTION_z OPTION_R "k:a:tT:h", long_options, NULL)) != -1) {
         switch (opt) {
             case 'c':
```

### Build examples

```shell
$ ./examples/buildall.py
```

### Run server/client examples

Pings don't return.

#### Server
```shell
$ ./build/examples/csp_server -c vcan0 -a 1
Initialising CSP
INIT CAN: device: [vcan0], bitrate: 1000000, promisc: 1
Connection table
[00 0x63375bb10c20] S:0, 0 -> 0, 0 -> 0 (17) fl 0
[01 0x63375bb10d30] S:0, 0 -> 0, 0 -> 0 (18) fl 0
[02 0x63375bb10e40] S:0, 0 -> 0, 0 -> 0 (19) fl 0
[03 0x63375bb10f50] S:0, 0 -> 0, 0 -> 0 (20) fl 0
[04 0x63375bb11060] S:0, 0 -> 0, 0 -> 0 (21) fl 0
[05 0x63375bb11170] S:0, 0 -> 0, 0 -> 0 (22) fl 0
[06 0x63375bb11280] S:0, 0 -> 0, 0 -> 0 (23) fl 0
[07 0x63375bb11390] S:0, 0 -> 0, 0 -> 0 (24) fl 0
Interfaces
LOOP       addr: 0 netmask: 5 dfl: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

CAN        addr: 1 netmask: 0 dfl: 1
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

Server task started
```

#### Client
```shell
$ ./build/examples/csp_client -c vcan0 -a 2 -C 1
Initialising CSP
INIT CAN: device: [vcan0], bitrate: 1000000, promisc: 1
Connection table
[00 0x55d72e660be0] S:0, 0 -> 0, 0 -> 0 (17) fl 0
[01 0x55d72e660cf0] S:0, 0 -> 0, 0 -> 0 (18) fl 0
[02 0x55d72e660e00] S:0, 0 -> 0, 0 -> 0 (19) fl 0
[03 0x55d72e660f10] S:0, 0 -> 0, 0 -> 0 (20) fl 0
[04 0x55d72e661020] S:0, 0 -> 0, 0 -> 0 (21) fl 0
[05 0x55d72e661130] S:0, 0 -> 0, 0 -> 0 (22) fl 0
[06 0x55d72e661240] S:0, 0 -> 0, 0 -> 0 (23) fl 0
[07 0x55d72e661350] S:0, 0 -> 0, 0 -> 0 (24) fl 0
Interfaces
LOOP       addr: 0 netmask: 5 dfl: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

CAN        addr: 2 netmask: 0 dfl: 1
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

Client started
Ping address: 1, result -1 [mS]
reboot system request sent to address: 1
Ping address: 1, result -1 [mS]
reboot system request sent to address: 1
Ping address: 1, result -1 [mS]
reboot system request sent to address: 1
```

yashi, 2024-10-15:
Indeed a CAN interface doesn't work on my system as well.

OTOH, a USART interface seems to work:

**Server**
```
❯ ./builddir/examples/csp_server -k /tmp/pty1 -a 1
Initialising CSP
Connection table
[00 0x7f2c5e6408c0] S:0, 0 -> 0, 0 -> 0 (17) fl 0
[01 0x7f2c5e6409d8] S:0, 0 -> 0, 0 -> 0 (18) fl 0
[02 0x7f2c5e640af0] S:0, 0 -> 0, 0 -> 0 (19) fl 0
[03 0x7f2c5e640c08] S:0, 0 -> 0, 0 -> 0 (20) fl 0
[04 0x7f2c5e640d20] S:0, 0 -> 0, 0 -> 0 (21) fl 0
[05 0x7f2c5e640e38] S:0, 0 -> 0, 0 -> 0 (22) fl 0
[06 0x7f2c5e640f50] S:0, 0 -> 0, 0 -> 0 (23) fl 0
[07 0x7f2c5e641068] S:0, 0 -> 0, 0 -> 0 (24) fl 0
Interfaces
LOOP       addr: 0 netmask: 5 dfl: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

KISS       addr: 1 netmask: 0 dfl: 1
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

Server task started
Packet received on SERVER_PORT: Hello world A
Packet received on SERVER_PORT: Hello world B
Packet received on SERVER_PORT: Hello world C
Packet received on SERVER_PORT: Hello world D
Packet received on SERVER_PORT: Hello world E
Packet received on SERVER_PORT: Hello world F
Packet received on SERVER_PORT: Hello world G
Packet received on SERVER_PORT: Hello world H
Packet received on SERVER_PORT: Hello world I
Packet received on SERVER_PORT: Hello world J
```

**Client**
```
❯ ./builddir/examples/csp_client -k /tmp/pty2 -a 2 -C 1
Initialising CSP
Connection table
[00 0x7f61a79888c0] S:0, 0 -> 0, 0 -> 0 (17) fl 0
[01 0x7f61a79889d8] S:0, 0 -> 0, 0 -> 0 (18) fl 0
[02 0x7f61a7988af0] S:0, 0 -> 0, 0 -> 0 (19) fl 0
[03 0x7f61a7988c08] S:0, 0 -> 0, 0 -> 0 (20) fl 0
[04 0x7f61a7988d20] S:0, 0 -> 0, 0 -> 0 (21) fl 0
[05 0x7f61a7988e38] S:0, 0 -> 0, 0 -> 0 (22) fl 0
[06 0x7f61a7988f50] S:0, 0 -> 0, 0 -> 0 (23) fl 0
[07 0x7f61a7989068] S:0, 0 -> 0, 0 -> 0 (24) fl 0
Interfaces
LOOP       addr: 0 netmask: 5 dfl: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

KISS       addr: 2 netmask: 0 dfl: 1
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

Client started
Ping address: 1, result 0 [mS]
reboot system request sent to address: 1
Ping address: 1, result 0 [mS]
reboot system request sent to address: 1
Ping address: 1, result 0 [mS]
reboot system request sent to address: 1
Ping address: 1, result 0 [mS]
reboot system request sent to address: 1
Ping address: 1, result 0 [mS]
reboot system request sent to address: 1
Ping address: 1, result 0 [mS]
reboot system request sent to address: 1
Ping address: 1, result 0 [mS]
reboot system request sent to address: 1
Ping address: 1, result 0 [mS]
reboot system request sent to address: 1
Ping address: 1, result 0 [mS]
reboot system request sent to address: 1
Ping address: 1, result 0 [mS]
```

Would you mind checking it?

takumiando, 2024-10-16:
Yes. Currently, this bug happens only with CAN interface with CSP v1.
However, I think we need to choose one of the following options.

- Remove union from ```csp_packet_t```(https://github.com/libcsp/libcsp/pull/710)
- Remove ```rx_timestamp = 0``` from ```csp_id_strip()```
  - We need to keep it for the future? https://github.com/libcsp/libcsp/issues/531

yashi, 2024-10-16:
> Yes. Currently, this bug happens only with CAN interface with CSP v1.

Would you mind explaining why?

takumiando, 2024-10-16:
Sorry, my explanation was wrong.

>OTOH, a USART interface seems to work:

- A USART interface doesn't use ```cfpid```, so this bug doesn't happen.
- A CAN interface with CSP v2 calls ```csp_id_strip()``` in the end of receiving, and this bug doesn't happen.
- A CAN interface with CSP v1 calls ```csp_id_strip()``` in the begin of receiving,  and this bug **happens**.

Is the timing of calling ```csp_id_strip()``` in CAN CSP v1 wrong?

PR #717 (2024-10-31): csp_packet: Remove unused timestamp_rx
Remove the unsued member variable `timetamp_rx` from the struct `csp_packet_s`.

In #531 @johandc says that it will be used for GPS timesync but it's not upstreamed yet. We can re-introduce when the feature is ready.

Fixes #708
closed 2024-10-31 by yashi