libcsp/libcsp #713: Can't compile CSP1.0 with UART

poomravint, 2024-10-22:
I just installed CSP1.0 then I try to compile csp_server_client.c in examples, I use this command to compile :  gcc csp_server_client.c -lcsp -o csp
Then I got this : 
[image]
How to solve this problems ?

yashi, 2024-10-22:
Do you really need to use libcsp v1? Or do you just need CSP protocol version 1? If latter, please use the libcsp v2 or `develop` branch. Libcsp v1.0 is very old and not supported any more.

poomravint, 2024-10-24:
> Do you really need to use libcsp v1? Or do you just need CSP protocol version 1? If latter, please use the libcsp v2 or `develop` branch. Libcsp v1.0 is very old and not supported any more.
> 
> There is a PR to activate protocol version 1 in the test. #709
> 
> This PR is good but it's just waiting for other bug to be fixed.
> 
> Hope this helps.

> Do you really need to use libcsp v1? Or do you just need CSP protocol version 1? If latter, please use the libcsp v2 or `develop` branch. Libcsp v1.0 is very old and not supported any more.
> 
> There is a PR to activate protocol version 1 in the test. #709
> 
> This PR is good but it's just waiting for other bug to be fixed.
> 
> Hope this helps.

Yes I need to use libcsp v1 because my team are developing module for Cubesat then that Cubesat use libcsp v1 for communication protocol but Actually we will use CSP over CAN.
I just want to try to use CSP over UART if I can't fix this problem , I will use CSP over CAN.

Thanks for your reply

yashi, 2024-10-24:
There are two things we version.

- libcsp -- the library
- CSP protocol -- the protocol

If you say _I need to use libcsp v1_, this means:

```
❯ git checkout v1.0
```

But I doubt it because you also says *Cubesat use libcsp v1 for communication protocol*.  The communication depends on the protocol version, not the library version.  As I said libcsp v2, the library, supports both protocol version 2 and 1.

poomravint, 2024-10-24:
How can I know protocol version and library version. I clone your libcsp from libcsp-1 branch what is protocol version and library version from this branch ?

yashi, 2024-10-24:
if you checkout `libcsp-1`, do:

```
❯ git describe
v1.6-13-g7fba4018
```

This means you are using v1.6 with 13 more commits on top of it.  See https://git-scm.com/docs/git-describe

To set protocol version, read #709.

poomravint, 2024-10-25:
If I want to use CSP protocol 1 ,Which libcsp version I have to use ?
I am sorry If you have already answered my question but I don't understand.

yashi, 2024-10-25:
> If I want to use CSP protocol 1 ,Which libcsp version I have to use ?

You can use any libcsp version including `develop` branch.

> How I know what protocol version that I use

You must know what protocol version *you* want to use.

> Do I have to config it or not ?

Yes. Set `csp_conf.version` to `1`.

```c

int main()
{
    csp_conf.version = 1;
    csp_init();
        :
}

poomravint, 2024-10-28:
Thanks I got it.

closed 2024-10-28 by yashi

poomravint, 2024-10-29:
Can I ask you more question
Now I am coding Sender and Receiver communication by CAN
If I want to protocol version 1 just add : `csp_conf.version = 1;`  before `csp_init(); ` right ? or  I have to add csp_conf in some function ?
then `csp_conf.version = 1;`  I have to add it on sender or receiver or both

poomravint, 2024-10-29:
Can I ask you any question
Now I use CSP over CAN
Sender.c and Receiver.c
But When I set csp_conf.version = 1; both of Sender and Reciever
Reciever can't accept a connection
but when I change csp_conf.version = 2; both of Sender and Reciever
Reciever can accept a connection
I don't know what happen
this is my code [my code](https://github.com/poomravint/CSP-CAN-on-RasPI-)
My code on Github I haven't add `csp_conf.version` but the code that I run ,I have already add `csp_conf.version`

yashi, 2024-10-29:
See #708, #695, and #531

yashi, 2024-11-01:
@poomravint The issue should have been fixed. Please try again and let us know how it goes.

poomravint, 2024-11-05:
Sorry for late reply I just try to fix it.
Now I can use CAN with CSP Protocol 1.0
I deleted union from `csp_packet_t` and It's work.
[image]

poomravint, 2024-11-07:
I just clone libcsp new version then I don't find my problems. Thanks