libcsp/libcsp #720: LibCSP on bare metal without RTOS

ReneCanadian, 2024-11-06:
Hi, I wanted to ask if there is any sensible way to use LibCSP on a device without RTOS? I couldn't seem to find information about that in the documentation, and other people from the industry seem to think this couldn't really be done.

 In general I'm using an MSP430FR69791, which is pretty low on resources 2kB RAM and 64 kB of FRAM. I'm not sure if diving into trying to write a port is worth doing and I was hoping that maybe someone already tried something like this.
 
 I'd appreciate any help or opinion on the matter, thanks!

dimitrovand, 2024-11-06:
Hello! I don't think its meant to work without RTOS. The topic is really interesting so you can checkout https://github.com/dimitrovand/libcsp/tree/baremetal and let me know if it works for you. Note that this is an implementation for Linux but without using any OS specifics and thus you won't be able to use RDP mode.

ReneCanadian, 2024-11-07:
Thanks! Will do!

yashi, 2024-11-27:
I'm closing this for now. Reopen or open new one if you have more questions.