libcsp/libcsp #721: How to use csp_sfp_send()?

hernaangq, 2024-11-06:
Hello,

Does anyone have an example of how to use **csp_sfp_send()**?
I am currently trying to send 1000 bytes but I am only receiving 5 CSP packets, I am pretty sure does not how it is suppose to work.

Thanks in advance.
Hernán

jganser, 2024-11-07:
With `csp_sfp_send()` you set the MTU value this value indicates how many bytes at maximum can be part of one sfp packet. Although csp_packet has 256 bytes usually some are necessary for sfp headers as far as i understood it.
So long story short if your using `mtu >= 200` than it makes totally sense that 1000 bytes are send in 5 packets.
As 5 chunks each of a size bigger or equal to 200 bytes are in sum 1000 bytes or more.

For full reference:
```c
/**
 * Send data over a CSP connection.
 *
 * Uses csp_sfp_send_own_memcpy() with standard memcpy().
 *
 * @param[in] conn established connection for sending SFP packets.
 * @param[in] data data to send
 * @param[in] datasize  size of \a data
 * @param[in] mtu  maximum transfer unit (bytes), max data chunk to send.
 * @param[in] timeout unused as of CSP version 1.6
 * @return #CSP_ERR_NONE on success, otherwise an error.
 */
static inline int csp_sfp_send(csp_conn_t * conn, const void * data, unsigned int datasize, unsigned int mtu, uint32_t timeout)
``` 

On a side note I am looking into sending pictures currently via sfp, but am stuck in the csp_buffer limitations... The message fails because it runs out of buffers to allocate, it seems.

jganser, 2024-11-07:
My issue was in the zmq_task of the zmq_interface i was using, as it was reading messages and allocating new packets faster than sfp was freeing them...

My hotfix is to introduce a 1ms sleep...:

```c
void * csp_zmqhub_task(void * param) {

	zmq_driver_t * drv = param;
	csp_packet_t * packet;
	const uint32_t HEADER_SIZE = (csp_conf.version == 2) ? 6 : 4;

	while (1) {
		int ret;
		(void)ret; /* Silence unused variable warning (promoted to an error if -Werr) issued when building with NDEBUG (release with asserts turned off) */
		zmq_msg_t msg;

		ret = zmq_msg_init_size(&msg, sizeof(packet->data) + HEADER_SIZE);
		assert(ret == 0);

		// Receive data
		if (zmq_msg_recv(&msg, drv->subscriber, 0) < 0) {
			csp_print("ZMQ RX err %s: %s\n", drv->iface.name, zmq_strerror(zmq_errno()));
			continue;
		}

		unsigned int datalen = zmq_msg_size(&msg);
		if (datalen < HEADER_SIZE) {
			csp_print("ZMQ RX %s: Too short datalen: %u - expected min %u bytes\n", drv->iface.name, datalen, HEADER_SIZE);
			zmq_msg_close(&msg);
			continue;
		}

                sleep(0.001); // ugly hotfix for sfp...

		// Create new csp packet
		packet = csp_buffer_get(0);
		if (packet == NULL) {
			csp_print("RX %s: Failed to get csp_buffer(%u)\n", drv->iface.name, datalen);
			zmq_msg_close(&msg);
			continue;
		}

		// Copy the data from zmq to csp
		const uint8_t * rx_data = zmq_msg_data(&msg);

		csp_id_setup_rx(packet);

		memcpy(packet->frame_begin, rx_data, datalen);
		packet->frame_length = datalen;

		/* Parse the frame and strip the ID field */
		if (csp_id_strip(packet) != 0) {
			drv->iface.rx_error++;
			csp_buffer_free(packet);
		    zmq_msg_close(&msg);
			continue;
		}

		// Route packet
		csp_qfifo_write(packet, &drv->iface, NULL);

		zmq_msg_close(&msg);
	}

	return NULL;
}
```

jganser, 2024-11-14:
which version of the lib do you use?

the signature of `static inline int csp_sfp_recv(csp_conn_t * conn, void ** dataout, int * datasize, uint32_t timeout)` doesn't seem to conform with your csp_sfp_recv call: `csp_sfp_recv(conn_sfp_handler, (void**)&dataout, &datasize, 500, 10000);`

Do you get any other error messages beside the return code of -103 which stands for: `CSP_ERR_SFP`
Which happens if either:
- the sfp header is not there (which gets appended at the end of the data (which is the reason why the mtu has to be smaller than 255 by the size of the header in the standard config of libcsp)
- the offset that is mentioned in the sfp_header doesn't match the one that we currently calculated, meaning that we got a compromised packet or missed one (or many)
- datasize and totalsize don't match
- we got an packet without length (empty packet)

All these errors have the same error code...
Therefor my recommendation is to compile the library with the lines uncommented that do the printing inside each of these error checks to get to know what exactly is your issue.

Hope this is a little helpful to you.

jganser, 2024-11-14:
For further inspiration check out: 

- https://github.com/jganser/zmq_sfp_conn_test
- https://github.com/libcsp/libcsp/issues/723

yashi, 2024-11-27:
I'm closing this for now. Reopen or open new one if you have more questions.

closed 2024-11-27 by yashi