libcsp/libcsp #723: SFP: ZMQ Task requesting new buffers faster than sfp was freeing old ones

jganser, 2024-11-07:
My issue was in the zmq_task of the zmq_interface i was using, as it was reading messages and allocating new packets faster than sfp was freeing them...

My hotfix is to introduce a 1ms sleep...:

```c
void * csp_zmqhub_task(void * param) {

	zmq_driver_t * drv = param;
	csp_packet_t * packet;
	const uint32_t HEADER_SIZE = (csp_conf.version == 2) ? 6 : 4;

	while (1) {
		int ret;
		(void)ret; /* Silence unused variable warning (promoted to an error if -Werr) issued when building with NDEBUG (release with asserts turned off) */
		zmq_msg_t msg;

		ret = zmq_msg_init_size(&msg, sizeof(packet->data) + HEADER_SIZE);
		assert(ret == 0);

		// Receive data
		if (zmq_msg_recv(&msg, drv->subscriber, 0) < 0) {
			csp_print("ZMQ RX err %s: %s\n", drv->iface.name, zmq_strerror(zmq_errno()));
			continue;
		}

		unsigned int datalen = zmq_msg_size(&msg);
		if (datalen < HEADER_SIZE) {
			csp_print("ZMQ RX %s: Too short datalen: %u - expected min %u bytes\n", drv->iface.name, datalen, HEADER_SIZE);
			zmq_msg_close(&msg);
			continue;
		}

                sleep(0.001); // ugly hotfix for sfp...

		// Create new csp packet
		packet = csp_buffer_get(0);
		if (packet == NULL) {
			csp_print("RX %s: Failed to get csp_buffer(%u)\n", drv->iface.name, datalen);
			zmq_msg_close(&msg);
			continue;
		}

		// Copy the data from zmq to csp
		const uint8_t * rx_data = zmq_msg_data(&msg);

		csp_id_setup_rx(packet);

		memcpy(packet->frame_begin, rx_data, datalen);
		packet->frame_length = datalen;

		/* Parse the frame and strip the ID field */
		if (csp_id_strip(packet) != 0) {
			drv->iface.rx_error++;
			csp_buffer_free(packet);
		    zmq_msg_close(&msg);
			continue;
		}

		// Route packet
		csp_qfifo_write(packet, &drv->iface, NULL);

		zmq_msg_close(&msg);
	}

	return NULL;
}
```

_Originally posted by @jganser in https://github.com/libcsp/libcsp/issues/721#issuecomment-2462338602_

---

A `magic number` sleep call is not my desired solution. Any ideas how to fix this kind of behavior in a sustainable way?

yashi, 2024-11-10:
Hi thanks you for taking time to report it. Would you please give us a complete [MRE](https://en.wikipedia.org/wiki/Minimal_reproducible_example) we can test?

Thanks.

jganser, 2024-11-13:
Thanks for answering. I'll gladly provide the MRE. You can find it in this [repository](https://github.com/jganser/zmq_sfp_conn_test) as a standalone build (with relative path to the local libcsp repository though).

In my last test with this a png with a size of ~17kb was received sometimes successful, but mostly it was not received.

Error was instead:

```
...
RX ZMQHUB: Failed to get csp_buffer(255)
RX ZMQHUB: Failed to get csp_buffer(255)
RX ZMQHUB: Failed to get csp_buffer(255)
Error in csp_sfp_recv: -103
...
```

The error comes multiple times, sometime with more `RX ZMQHUB: Failed to get csp_buffer(255)` lines sometimes with less.

libcsp build is from commit: 

55471e4bc6db75ffc09c27c13a1e8b35d00cfa78: csp_rdp: Prevent NULL pointer dereference

my OS: Ubuntu 24.04

If you need more details please notify me.

jganser, 2024-11-13:
Persists to exist with commit:

58559147: unittest: Rename "tests" directory to "unittests"

yashi, 2024-11-13:
Got it built with the following patch:

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 4aac31f..4e6303e 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -6,6 +6,7 @@ find_package(PkgConfig)
 
 pkg_search_module(LIBCSP libcsp)
 file(REAL_PATH "../libcsp/include" csp_inc)
+file(REAL_PATH "../libcsp/builddir/include" csp_inc2)
 
 add_executable(img_rcv csp_client_posix.c
   img.c
@@ -17,8 +18,8 @@ add_executable(img_sender csp_client_posix.c
   minimal_zmq_client.c
 )
 
-target_include_directories(img_rcv     PRIVATE ${csp_inc})
-target_include_directories(img_sender  PRIVATE ${csp_inc})
+target_include_directories(img_rcv     PRIVATE ${csp_inc} ${csp_inc2})
+target_include_directories(img_sender  PRIVATE ${csp_inc} ${csp_inc2})
 
 target_link_libraries(img_rcv    PRIVATE csp Threads::Threads)
 target_link_libraries(img_sender PRIVATE csp Threads::Threads)
```

Now how can I test it?

**img_rcv**
```
❯ ./builddir/img_rcv
Initialising CSP
Connection table
[00 0x7f4f53152080] S:0, 0 -> 0, 0 -> 0 (17) fl 0
[01 0x7f4f53152198] S:0, 0 -> 0, 0 -> 0 (18) fl 0
[02 0x7f4f531522b0] S:0, 0 -> 0, 0 -> 0 (19) fl 0
[03 0x7f4f531523c8] S:0, 0 -> 0, 0 -> 0 (20) fl 0
[04 0x7f4f531524e0] S:0, 0 -> 0, 0 -> 0 (21) fl 0
[05 0x7f4f531525f8] S:0, 0 -> 0, 0 -> 0 (22) fl 0
[06 0x7f4f53152710] S:0, 0 -> 0, 0 -> 0 (23) fl 0
[07 0x7f4f53152828] S:0, 0 -> 0, 0 -> 0 (24) fl 0
Interfaces
LOOP       addr: 0 netmask: 14 dfl: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

ZMQHUB     addr: 1 netmask: 0 dfl: 1
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

IMG RCV Client task started
csp_accept: timeout
csp_accept: timeout
csp_accept: timeout
csp_accept: timeout
```

**img_sender**
```
❯ ./builddir/img_sender
Initialising CSP
Connection table
[00 0x7f1f481ef080] S:0, 0 -> 0, 0 -> 0 (17) fl 0
[01 0x7f1f481ef198] S:0, 0 -> 0, 0 -> 0 (18) fl 0
[02 0x7f1f481ef2b0] S:0, 0 -> 0, 0 -> 0 (19) fl 0
[03 0x7f1f481ef3c8] S:0, 0 -> 0, 0 -> 0 (20) fl 0
[04 0x7f1f481ef4e0] S:0, 0 -> 0, 0 -> 0 (21) fl 0
[05 0x7f1f481ef5f8] S:0, 0 -> 0, 0 -> 0 (22) fl 0
[06 0x7f1f481ef710] S:0, 0 -> 0, 0 -> 0 (23) fl 0
[07 0x7f1f481ef828] S:0, 0 -> 0, 0 -> 0 (24) fl 0
Interfaces
LOOP       addr: 0 netmask: 14 dfl: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

ZMQHUB     addr: 1 netmask: 0 dfl: 1
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

Error: Cannot open file /home/johannes/Pictures/Espresso_ViCAFE_BrewGuide.png
Error reading file: /home/johannes/Pictures/Espresso_ViCAFE_BrewGuide.png.
```

Can the png file be anything?

yashi, 2024-11-13:
OK, with `zmqproxy` running and specifing some random png file, it changed a bit:

```
❯ ./builddir/img_rcv
Initialising CSP
Connection table
[00 0x7f3c1da4a080] S:0, 0 -> 0, 0 -> 0 (17) fl 0
[01 0x7f3c1da4a198] S:0, 0 -> 0, 0 -> 0 (18) fl 0
[02 0x7f3c1da4a2b0] S:0, 0 -> 0, 0 -> 0 (19) fl 0
[03 0x7f3c1da4a3c8] S:0, 0 -> 0, 0 -> 0 (20) fl 0
[04 0x7f3c1da4a4e0] S:0, 0 -> 0, 0 -> 0 (21) fl 0
[05 0x7f3c1da4a5f8] S:0, 0 -> 0, 0 -> 0 (22) fl 0
[06 0x7f3c1da4a710] S:0, 0 -> 0, 0 -> 0 (23) fl 0
[07 0x7f3c1da4a828] S:0, 0 -> 0, 0 -> 0 (24) fl 0
Interfaces
LOOP       addr: 0 netmask: 14 dfl: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

ZMQHUB     addr: 1 netmask: 0 dfl: 1
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

IMG RCV Client task started
RX ZMQHUB: Failed to get csp_buffer(255)
got connection
RX ZMQHUB: Failed to get csp_buffer(255)
Error in csp_sfp_recv: -103
csp_accept: timeout
RX ZMQHUB: Failed to get csp_buffer(255)
got connection
Error in csp_sfp_recv: -103
csp_accept: timeout
RX ZMQHUB: Failed to get csp_buffer(255)
got connection
Error in csp_sfp_recv: -103
csp_accept: timeout
RX ZMQHUB: Failed to get csp_buffer(255)
got connection
Error in csp_sfp_recv: -103
```

Just to make sure, `RX ZMQHUB: Failed to get csp_buffer(255)` is what you are refering?

yashi, 2024-11-13:
> Just to make sure, `RX ZMQHUB: Failed to get csp_buffer(255)` is what you are refering?

If so, that's expected I supporse. `csp_zmqhub_task()` is the one consuming the buffer, which is created as free running thread. And it goes:

```c
void * csp_zmqhub_task(void * param) {
    :

	while (1) {
    :
		zmq_msg_recv(&msg, drv->subscriber, 0) < 0)
    :
		packet = csp_buffer_get(0);
		if (packet == NULL) {
			csp_print("RX %s: Failed to get csp_buffer(%u)\n", drv->iface.name, datalen);
			zmq_msg_close(&msg);
			continue;
		}
    :
		// Route packet
		csp_qfifo_write(packet, &drv->iface, NULL);
    :
	}

	return NULL;
}
```

So it just allocate buffer as it receive data.

I'm not saying it's a good piece of code. But that's what it does right now.

The current implementation of libcsp receives as much as it can get buffer.  If it can't dump the message, close connection, and try again.  Thus, if the incoming message comes fast, you might see the message unless thread priority of other thread is higher and RX task wan't get run. In that case, Linux kernel will cache the ZeroMQ packet in the kernel as much as the kernel is allow to cache. If the kernel failed to obtain memory to hold the incomming packet, it will a) dump the packet if it's UDP, or b) do trafic control on TCP.

jganser, 2024-11-14:
You are absolutely right, that is the behavior in the current implementation. Which is quite problematic with SFP, as SFP fails if just one packet is thrown away as far as i understood it.

- What would you recommend to solve this?
- Is RDP an option for this problem?
- Would a semaphore based guard for the packet acquisition make sense?
- Or should we think of delaying the SFP send function to allow the other site some time to receive it?

I guess this problem could in theory arise on other interfaces as well, as long as the sending is faster than the receiving function in SFP...

Please let me know if you are willing to change something in the implementation or desire a specific kind of solution, I am willing to help with implementing it, as long as it is desired.

jganser, 2024-11-14:
Ok, just tested it with a RDP connection locally and it worked for me. So unless RDP is not an option, my problem is solved.

yashi, 2024-11-15:
> * What would you recommend to solve this?

Actually, I don't know. If you don't have enough memory to hold _any_ data you receive, you either have to
- Drop the received data on the floor hoping that the sender will send it again when memory is available (RDP), or
- Overwrite some other data in order to hold the received data

These are the only option I can imagine.

> * Is RDP an option for this problem?

> Ok, just tested it with a RDP connection locally and it worked for me.

I seem it worked for you?

> * Would a semaphore based guard for the packet acquisition make sense?

What do you expect when consumer side is holding a semaphore?

> * Or should we think of delaying the SFP send function to allow the other site some time to receive it?

This only helps for transmission jitters.  The delay on the sender side is good as long as the whole sending side (in the view of sender) or receiving side (in the view of receiver) data throughput is lower than consuer (receiver with SFP retrieval side)  throughput. If sending throughput is higher than consumer, it will eventually run out of memory.

> I guess this problem could in theory arise on other interfaces as well, as long as the sending is faster than the receiving function in SFP...

right.

> Please let me know if you are willing to change something in the implementation or desire a specific kind of solution, I am willing to help with implementing it, as long as it is desired.

I'd love to do it if there is a practical solution for it. I don't have it ATM.