libcsp/libcsp #724: libcsp-1 backwards compatibility

n-log-gos, 2024-11-07:
## Description

How can I achieve compatibility between a _node A_ using **libcsp-2.0 + csp_conf.version=1**  and a _node B_ that uses **libcsp-1** ? I can't access the source code of _node B_ so upgrading isn't possible. 

`csp_packet_t` structure is inherently different between each version, the most prominent difference being the fixed big-endianness of the header of the newer versions. Is there an effort already done to create this compatibility?

## Steps to Reproduce

Compile a CSP server and client from libcsp-2.0 and libcsp-1, communication will fail.

## Environment Details

Provide information about your environment, including:

- Build OS:  Ubuntu 24.04
- Target OS: Ubunut 24.04
- libcsp commit hash: f74ee417 vs. 7fba4018

yashi, 2024-11-10:
Does #713 help?

n-log-gos, 2024-11-11:
> Does #713 help?

I have been using `csp_conf.version=1` - via the `-v` option referenced in the PR. However, this does not entirely work, as what I need is to use a "libcsp-2.0 node" (with `-v1` option to set CSP version 1) and a "libcsp-1" node, i.e., a propietary software which source code / upgrade I can't access to. 

My questions is: is there a test mode where I could set libcsp-2 to be compatible with libcsp-1 nodes? I.e., nodes that are compatible with the old, libcsp-1 `csp_packet_t` struct ? Or should I downgrade and just use libcsp-1 for all my nodes ?

yashi, 2024-11-11:
We also have a Cubesat component runing CSPv1. And we have successfully communicating using `develop` branch of libcsp in CSP protocol v1.

yashi, 2024-11-11:
Are you able to capture the packets using Wireshark? Does https://github.com/KantaTamura/CSP-dissector help?

n-log-gos, 2024-11-11:
> We also have a Cubesat component runing CSPv1. And we have successfully communicating using `develop` branch of libcsp in CSP protocol v1.

Is that Cubesat component built using libcsp-1 library? Or is built against libcsp2.0, and using `csp_conf = 1 `?

yashi, 2024-11-11:
I’m not sure; it’s not disclosed. However, we bought it long before we tagged v2, so it must be running libcsp-v1.

I just checked out the `libcsp-v1` tag and tested it against the `develop` branch with protocol v1—it seems to work fine. Have you tried this?

yashi, 2024-11-11:
Ah, ZMQ....

Is your _Node B_ using ZMQ? On Linux? Why can't you update it?

n-log-gos, 2024-11-11:
> Is your _Node B_ using ZMQ? On Linux? Why can't you update it?
Yes, it uses ZMQ proxy interface on Linux Ubuntu 24.04. I can't update it because I don't have access to the source code of that software.

yashi, 2024-11-17:
Would you mind creating a PR for it?