libcsp/libcsp #725: How to set CSP header flags ?

poomravint, 2024-11-11:
I use CRC and HMAC on the sender :
` printf("CRC Sucess : %d\n", csp_crc32_append(packet));
  printf("HMAC Sucess : %d\n", csp_hmac_append(packet, false));`
and set bit flags :
`packet->id.flags = (CSP_FHMAC | CSP_FCRC32);`
    
But when the receiver get csp packet from sender flags = 0;
`printf("Priority : %d\n", req->id.pri);
 printf("Src : %d\n", req->id.src);
 printf("DST : %d\n", req->id.dst);
 printf("FLAG : %d\n", req->id.flags);`
[image]

yashi, 2024-11-11:
This is what I got:

```
Priority : 2
Src : 2
DST : 1
FLAG : 9
```

With:

```diff
diff --git a/examples/csp_server.c b/examples/csp_server.c
index eff0d812..2aaa7e09 100644
--- a/examples/csp_server.c
+++ b/examples/csp_server.c
@@ -65,6 +65,10 @@ void server(void) {
 			case SERVER_PORT:
 				/* Process packet here */
 				csp_print("Packet received on SERVER_PORT: %s\n", (char *) packet->data);
+				csp_print("Priority : %d\n", packet->id.pri);
+				csp_print("Src : %d\n", packet->id.src);
+				csp_print("DST : %d\n", packet->id.dst);
+				csp_print("FLAG : %d\n", packet->id.flags);
 				csp_buffer_free(packet);
 				++server_received;
 				break;
diff --git a/samples/posix/simple-send-usart/src/main.c b/samples/posix/simple-send-usart/src/main.c
index 745cfec4..e8394252 100644
--- a/samples/posix/simple-send-usart/src/main.c
+++ b/samples/posix/simple-send-usart/src/main.c
@@ -34,7 +34,7 @@ int main(int argc, char * argv[])
 	iface->is_default = 1;
 
 	/* connect */
-	conn = csp_connect(CSP_PRIO_NORM, SERVER_ADDR, SERVER_PORT, 1000, CSP_O_NONE);
+	conn = csp_connect(CSP_PRIO_NORM, SERVER_ADDR, SERVER_PORT, 1000, CSP_O_CRC32 | CSP_O_HMAC);
 	if (conn == NULL) {
 		csp_print("Connection failed\n");
 		return 1;
```

The sample code is available in #726.

poomravint, 2024-11-12:
Okay I got it because this `conn = csp_connect(CSP_PRIO_NORM, SERVER_ADDR, SERVER_PORT, 0, CSP_O_NONE);`
I fill `CSP_O_NONE ` on the sender side. when my receiver got the message, FLAG = 0.
Thanks

closed 2024-11-12 by poomravint