libcsp/libcsp #730: How to use header in Hmac calculation

poomravint, 2024-11-12:
I try to use header in Hmac calculation by this `printf("HMAC Success : %d\n", csp_hmac_append(packet, true));`
But when I run my code I got : `Segmentation fault`.

yashi, 2024-11-12:
as always, please post with a [MRE](https://en.wikipedia.org/wiki/Minimal_reproducible_example). It's hard to see what's going on without it.

poomravint, 2024-11-12:
Ok I got it , I am so sorry.
This is my code :
[image]
and this is my result :
[image]

yashi, 2024-11-12:
please don't use screenshot. Just paste your code here or use [gist](https://gist.github.com/).

poomravint, 2024-11-12:
I am sorry this is my code : <script src="https://gist.github.com/poomravint/807ae77246054ed9344fd939628517f4.js"></script>

poomravint, 2024-11-12:
Ok This is my code : [https://gist.github.com/poomravint/807ae77246054ed9344fd939628517f4](url)

poomravint, 2024-11-12:
Ok This is my code : https://gist.github.com/poomravint/807ae77246054ed9344fd939628517f4

yashi, 2024-11-12:
OK I guess it's my fault. you need to call `csp_id_prepend(packet);` to initialize `frame_length`. Here is my MRE.

```c
#include <csp/csp.h>
#include <csp/csp_debug.h>
#include <csp/csp_id.h>
#include <csp/crypto/csp_hmac.h>

int main(int argc, char * argv[])
{
	csp_packet_t * packet;

	csp_init();

	packet = csp_buffer_get_always();
	csp_id_prepend(packet);

	csp_hmac_append(packet, true);
	csp_hmac_verify(packet, true);

	return 0;
}
```

poomravint, 2024-11-12:
Thanks Now I can send but I think my receiver have something wrong because my receiver got the correct data but when I use `printf("HMAC Result %d\n", csp_hmac_verify(req, true));`  then I got :
[image]
but the sender before I send to receiver I got check by `printf("HMAC Result %d\n", csp_hmac_verify(packet, true));`  then I got : 
[image]

I will try to fix it myself if I can't solve my problem I will ask you again Thanks a lot.

yashi, 2024-11-12:
Please open new issue when you can't figure it out. I'm closing this for now.

closed 2024-11-12 by yashi

poomravint, 2024-11-12:
Ok I got it , Thanks again