libcsp/libcsp #732: RTEMS port

luuke, 2024-11-12:
Hello,

Are you aware of any existing port of CSP for RTEMS?
I've been searching, but haven't had any luck finding one.

Best regards,
Łukasz

yashi, 2024-11-12:
Not that I know of.  Somebody else might know.

It'd be nice to have RTEMS port.

luuke, 2024-11-13:
Alright.

Then maybe you have a list of code modifications required to make a full port?
Something like:
- add something in CMakeLists
- add something in wscript
- add RTEMS specifics in src/arch
and so on. Just don't want to miss anything.

I don't want to make any promises, but I might give it a try.

yashi, 2024-11-13:
Porting the code under `src/arch` should be sufficient. You can choose one of the build systems we support (Waf, Meson, CMake), and I can help with the others. Of course, you can port to all three by yourself if you prefer. :smiley: 

One thing I'd like to request is a simple build test. If you’re familiar with GitHub Actions, please write a new action to build against RTEMS. If not, please let us know how to build it, so we can create a GitHub Action for it.

yashi, 2024-11-27:
I'm closing this for now. Reopen or open new one if you have more questions.

closed 2024-11-27 by yashi