libcsp/libcsp #733: csp_hmac_verify return HMAC failed

poomravint, 2024-11-14:
I have 2 code the first one is [Sender.c](https://gist.github.com/poomravint/859ebb4966e268358106e196cd4a3626) and the second one is [Receiver.c](https://gist.github.com/poomravint/afeae3c6a92d29d2cea98a83da171c37)
I use HMAC include header in hmac calculation `csp_hmac_append(packet, true)` in Sender
But when the Receiver got the message and use `csp_hmac_verify(req, true)` I got -100.
```HMAC Result = -100```
I think because before Sender send packet to Receiver the header packet is empty when I call `csp_hmac_append` so hmac calculation calculate the empty header
``` ROW 0 : 00
ROW 1 : 00
ROW 2 : 00
ROW 3 : 00
ROW 4 : 00
ROW 5 : 00
ROW 6 : 00
ROW 7 : 00
```
But when Receiver got the message and call `csp_hmac_verify` so hmac calculation calculate the header but header isn't empty
```ROW 0 : 00
ROW 1 : 00
ROW 2 : 00
ROW 3 : 00
ROW 4 : 84
ROW 5 : 12
ROW 6 : 92
ROW 7 : 09
```
that why when Receiver call `csp_hmac_verify(req, true)` then I got HMAC failed. But I have no idea to fix it.

moonlight83340, 2024-11-14:
Hi,
Waht libcsp version are you currently using ? 
If you use CSP_O_HMAC, current version will append the hmac :
https://github.com/libcsp/libcsp/blob/585591472db873bec6e4982051e21d97928fcb54/src/csp_io.c#L241

I'm not sure, DO you want hmac + CRC32  
https://gist.github.com/poomravint/859ebb4966e268358106e196cd4a3626#file-sender-c-L63

I haven't test yet your code, but you could try using CSP_O_HMAC only and not use hmac append on your sender part.
I will try to take some time to review your code.

I never used myself HMAC.

poomravint, 2024-11-14:
I using libcsp lasted version but use CSP protocol 1.0
I try to use CSP_O_HMAC only but it still not work . I got -100 from `csp_hmac_verify(req, true) `

moonlight83340, 2024-11-14:
If the packet was already verify by the app when received, it will strip the hmac part.
https://github.com/libcsp/libcsp/blob/develop/src/crypto/csp_hmac.c#L153
```
		/* Strip HMAC */
		packet->frame_length -= CSP_HMAC_LENGTH;
```

poomravint, 2024-11-15:
like you said If I use CSP_O_HMAC I don't need to use : https://gist.github.com/poomravint/859ebb4966e268358106e196cd4a3626#file-sender-c-L65
but When I use only CSP_O_HMAC and the receiver call `csp_hmac_verify` Why I got `-100 `that mean HMAC FAIL

poomravint, 2024-11-15:
Then If HMAC is fail how can I know

moonlight83340, 2024-11-15:
In libcsp route if HMAC failed, it will drop the packet. 
See here : 
https://github.com/libcsp/libcsp/blob/develop/src/csp_route.c#L78
The test is done in `csp_route_security_check` which is called in `csp_route_work`
https://github.com/libcsp/libcsp/blob/develop/src/csp_route.c#L109

So if you use the HMAC option (`CSP_O_HMAC`), if HMAC failed it will drop the packet.

poomravint, 2024-11-15:
Okay Thanks I got it

closed 2024-11-15 by poomravint