libcsp/libcsp #734: csp_buffer: Is it considered normal to have access to the data of previously used buffers?

dimitrovand, 2024-11-14:
## Description
Is it considered normal to have access to the data of previously used buffers?

## Steps to Reproduce
Run this code with default CSP settings:
```
#include <csp/csp.h>

int main() {
    csp_init();

    // use all buffer and free them
    for (unsigned int i = 0; i < CSP_BUFFER_COUNT; i++) {
        csp_packet_t * packet = csp_buffer_get_always();
        memset(packet->data, 0, sizeof(packet->data)); // clear buffer data
        memcpy(packet->data, "previous_data!!", i+1);  // put some data inside
        csp_buffer_free(packet);
    }

    // access the data of previously used buffers
    for (unsigned int i = 0; i < CSP_BUFFER_COUNT; i++) {
        csp_packet_t * packet = csp_buffer_get_always();
        csp_print("%lu: data=%s\n", (uintptr_t)packet, packet->data);
        csp_buffer_free(packet);
    }
}
```
## Expected behavior

Data, from previously used buffers, should not be visible.

## Actual Behavior

Data, from previously used buffers, is visible.

## Environment Details

- Build OS: any
- Target OS: any
- libcsp commit hash: any
- Meson version: any
- CMake version: any
- GNU Make version: any
- Ninja version: any

## Logs or console output from the very beginning to the end

## Additional Context
Output from provided code snippet:
```
126213084001520: data=p
126213084001856: data=pr
126213084002192: data=pre
126213084002528: data=prev
126213084002864: data=previ
126213084003200: data=previo
126213084003536: data=previou
126213084003872: data=previous
126213084004208: data=previous_
126213084004544: data=previous_d
126213084004880: data=previous_da
126213084005216: data=previous_dat
126213084005552: data=previous_data
126213084005888: data=previous_data!
126213084006224: data=previous_data!!
```

## Checklist

- [x] I've searched in "Issues" for similar issues, both open and closed.
- [x] I've used GitHub Flavored Markdown (GFM) to format my text.
- [x] I've avoided using screenshots for console output or logs
- [x] I've included all relevant details and steps to reproduce the issue.
- [x] I've provided information about my environment.
- [x] I've attached necessary logs or code snippet for minimal reproducible example.

yashi, 2024-11-14:
Hmm... What do you think? It's nice to clear the contents of a buffer before allocating. But that comes with a cost. In fact, if no MPU is involved, we can access anything anyway.

dimitrovand, 2024-11-15:
I'm not too concerned about embedded devices, since leaking data would require running external code on the system, which is unlikely. My bigger concern is when it's used as a shared library, though I'm not sure if that's even a valid concern at this point. Clearing the entire packet before each use could indeed be expensive. For now, some notes in the documentation might be sufficient, allowing users to decide how to handle it.

yashi, 2024-11-15:
I the buffer is shared with multiple processes on, say, Linux, it'd be a security concern. But, libcsp is userland library and it's impossible to share buffers with multiple processes with the current implementation.

Anyway, I'll finish up the PR (with a compile time switch and doc update) and ask you for your review, anyway.

closed 2024-11-21 by dimitrovand