libcsp/libcsp #736: tests: Should we add full coverage for at least one embedded system?

dimitrovand, 2024-11-16:
It seems there aren't any complete tests for embedded systems (beyond build tests) at the moment. Since CSP is primarily used on embedded devices (to my knowledge), adding full tests could be beneficial. The idea is straightforward: use a tool like [Renode](https://github.com/renode/renode) to simulate a complete embedded system. This would allow us to run all the tests we currently run on Linux in a simulated embedded environment, using an RTOS such as FreeRTOS or Zephyr.

yashi, 2024-11-16:
I'd love to!

Just to let others know that:

We have https://github.com/libcsp/libcsp-freertos but it's still targetting POSIX system. I don't use FreeRTOS and I've been asking users to provide how to build against MCUs without success.

We also have a test repository at https://github.com/spacecubics/libcsp-zephyr and it's also pushed to https://github.com/libcsp/libcsp-zephyr. We are very slow adding addint tests.

dimitrovand, 2024-11-18:
I can help out by providing FreeRTOS support in CMake and also adding tests for building CSP with FreeRTOS. I'm thinking of providing tests for building against STM32 since most them are fully emulated by Renode and I have access to such board.

yashi, 2024-11-18:
That's super! I'd love to have proper embedded MCU support with freertos.

If we can integrate freertos unit testing, we might want to move unit tests for Zephyr as well.

Are you going to use CMock? We are gonna have three unit tests frameworks but we can consolidate them later IMHO

dimitrovand, 2024-11-21:
If you have the time, let me know if something like [this](https://github.com/dimitrovand/csp_freertos_stm32) would do the job.
I haven't thought of unit testing yet, but its going to be the next step.

yashi, 2024-11-21:
Sure, whatever works for FreeRTOS developers is fine by me.  All I want to do is to run unit tests for FreeRTOS on GitHub Actions.

One thing I noticed is that you are using Renode. Is this required? Or is it just because it's easier to use?

dimitrovand, 2024-11-22:
If the goal is solely to run FreeRTOS unit tests, they likely don’t need to be executed on an embedded system. By unit tests, I’m referring to isolated tests like testing csp_rtable.h, which focuses on a linked list. Running such tests on embedded devices doesn’t seem to add significant value.
Renode is more suited for integration tests, such as end-to-end tests for ping or KISS/CAN interfaces, where the closest alternative to HIL is using a tool like Renode. That said, even Renode may not be strictly necessary for some cases, but ensuring CSP compiles correctly for a specific RTOS/OS is essential.
I chose Renode because it provides comprehensive support for boards and peripherals, is easy to use, and allows simple data exchange with the simulated system.

yashi, 2024-11-26:
Ah, OK. I misread in the first place. I wanted Unit Testing with a RTOS while you were thinking about Integration Testing. I agree that Renode is more suited for multi-node integration-like testing.

dimitrovand, 2024-11-29:
If its only for unit tests, I think building against POSIX ports for FreeRTOS and Zephyr should be enough. WDYT?

yashi, 2024-11-29:
If FreeRTOS users are OK with it, I'm all good.

dimitrovand, 2024-11-29:
I don't see any reason not to use it, since we're not aiming to test specific FreeRTOS or Zephyr ports.

closed 2024-12-04 by dimitrovand