libcsp/libcsp #740: Is the `OUTGOING_PORTS` check unused?

yashi, 2024-11-22:
> @johandc What was the intention of this block? Because `CSP_ID2_PORT_SIZE` is not available here, it's replaced to `0` in `#if` clause. Thus, it becomes `-CSP_CON_MAX > CSP_CON_MAX`, which is always false.> 
> > 
> You can check with the following code:> 
> > 
> ```c> 
> #define CSP_PORT_MAX_BIND 16> 
> > 
> #define OUTGOING_PORTS (((1 << (CSP_ID2_PORT_SIZE)) - 1) - CSP_PORT_MAX_BIND)> 
> #if OUTGOING_PORTS > CSP_CONN_MAX> 
> #error "More connections than available outgoing ports"> 
> #endif> 
> > 
> #include <stdio.h>> 
> > 
> int main()> 
> {> 
> #if OUTGOING_PORTS == -CSP_PORT_MAX_BIND> 
> 	printf("oops\n");> 
> #endif> 
> > 
> 	return 0;> 
> }> 
> ```> 
> > 
> ```> 
> gcc -Wall -Wextra -g a.c && ./a.out> 
> oops> 
> ```> 
> > 
> If `CSP_ID2_PORT_SIZE` is available, because it's defined as `6`, `OUTGOING_PORTS` will be `47`.  Then It's almost always false. 

 _Originally posted by @yashi in [589d76f](https://github.com/libcsp/libcsp/commit/589d76f1a62b61238275a7bff3fbdfd8300af924#r149386558)_

dimitrovand, 2024-11-29:
@yashi It's not used, but it should be. As you noticed, the comparison sign is inverted. Setting CSP_CONN_MAX above the current maximum (47) can lead to strange behavior.

yashi, 2024-12-02:
OK. Let me see what I can do. Thanks @dimitrovand as always.

closed 2026-08-18 by yashi