libcsp/libcsp #762: installation issue

panagiotispir, 2024-12-17:
## Description

I have a clean installation of ubuntu and I am trying to install and run the example. To have a good starting point, in order to build a beacon parcer.

I run:
sudo apt-get install cmake
sudo apt-get install ninja-build
sudo apt-get install build-essensials
git clone https://github.com/libcsp/libcsp.git
cmake -G Ninja -B builddir
cmake --build builddir
cmake --install builddir

and until here the steps implemented successfully.

when I trying to build examples using 
/buildall.py using cmake

i get
`-- Build files have been written to: /home/panos/libcsp/examples/build
ninja: Entering directory `build
[1/9] Linking C executable csp_server
FAILED: csp_server 
: && /usr/bin/cc  -rdynamic CMakeFiles/csp_posix_helper.dir/csp_posix_helper.o CMakeFiles/csp_server.dir/csp_server.o -o csp_server  -lcsp  -lcsp_common  -lcsp_common && :
/usr/bin/ld: cannot find -lcsp_common: No such file or directory
/usr/bin/ld: cannot find -lcsp_common: No such file or directory
collect2: error: ld returned 1 exit status
[2/9] Linking C executable csp_server_client
FAILED: csp_server_client 
: && /usr/bin/cc  -rdynamic CMakeFiles/csp_posix_helper.dir/csp_posix_helper.o CMakeFiles/csp_server_client.dir/csp_server_client.o -o csp_server_client  -lcsp  -lcsp_common  -lcsp_common && :
/usr/bin/ld: cannot find -lcsp_common: No such file or directory
/usr/bin/ld: cannot find -lcsp_common: No such file or directory
collect2: error: ld returned 1 exit status
[3/9] Linking C executable csp_arch
FAILED: csp_arch 
: && /usr/bin/cc  -rdynamic CMakeFiles/csp_arch.dir/csp_arch.o -o csp_arch  -lcsp  -lcsp_common && :
/usr/bin/ld: cannot find -lcsp_common: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/home/panos/libcsp/examples/./buildall.py", line 73, in <module>
    main(args.build_system)
  File "/home/panos/libcsp/examples/./buildall.py", line 61, in main
    build_with_cmake()
  File "/home/panos/libcsp/examples/./buildall.py", line 33, in build_with_cmake
    subprocess.check_call(cmake_compile)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', 'build']' returned non-zero exit status 1.`

moreover when thing to run the python_bindings_example_client.py I get no module name found libcsp_py3"

Is there any video about installation process or any step by step tutorial.

yashi, 2024-12-17:
[image]

yashi, 2024-12-17:
New one with `-GNinja` fix
[image]

yashi, 2024-12-17:
The whole steps
```
git clone https://github.com/libcsp/libcsp
cd libcsp
ls
git describe
sudo apt-get install cmake ninja-build libzmq3-dev libsocketcan-dev pkg-config check libyaml-dev
cmake --version
ninja --version
cmake -B builddir -DCSP_BUILD_SAMPLES=1 -GNinja && ninja -C builddir
echo '## test csp_arch'
./builddir/examples/csp_arch
echo '## test csp_server and client on vcan'
sudo ip link add dev vcan0 type vcan
sudo ip link set up vcan0
./builddir/examples/csp_server -c vcan0 -a 1
echo '## run ./builddir/examples/csp_client -c vcan0 -a 2 -C 1 on another terminal'
echo '## test csp_server and client on pty'
echo '## run socat -dd pty,raw,echo=0,link=/tmp/pty1 pty,raw,echo=0,link=/tmp/pty2 on another terminal'
./builddir/examples/csp_server -k /tmp/pty1 -a 1
echo '## run /builddir/examples/csp_client -k /tmp/pty2 -a 2 -C 1'
```

panagiotispir, 2024-12-17:
Thank you very much.
I think this video should also be uploaded to YouTube, to help other newbies, like me, with installation.

Moreover at this point the python bindings should also work?
I still have the error "no module name found libcsp_py3"

yashi, 2024-12-17:
You need `cmake ... -DCSP_ENABLE_PYTHON3_BINDINGS=1` for Python bindings.

panagiotispir, 2024-12-18:
Thanks again!
one more question,
I follow your instructions and I change
`cmake -B builddir -DCSP_BUILD_SAMPLES=1 -GNinja && ninja -C builddir`

to this
`cmake -B builddir -DCSP_BUILD_SAMPLES=1 -DCSP_ENABLE_PYTHON3_BINDINGS=1 -GNinja && ninja -C builddir`

for python bindings.

by this I mean expect to be able to use libcsp_py3 by the same way as for example pandas are installed and can be used on the system, but I still get "no module name found libcsp_py3" error.

yashi, 2024-12-19:
```
❯ rm -rf build && cmake -B build -DCSP_ENABLE_PYTHON3_BINDINGS=1 -DCSP_USE_RTABLE=1 && ninja -C build && PYTHONPATH=build python3 -c 'import libcsp_py3 as csp'
```

PR #767 (2024-12-22): doc: INSTALL: Add Python bindings build instructions for CMake
Document the steps to build Python bindings using CMake. Include necessary options and additional steps for enabling the routing table and setting `PYTHONPATH` for usage.

Also, add an additional headline level for improved section structure.

This fixes #762.
closed 2024-12-22 by yashi