libcsp/libcsp #779: CSP in Code Composer Studio

Aseke314, 2025-01-10:
Hello everyone. I am new to CSP. Can someone please explain how to use libcsp in Code Composer Studio project? I tried to compile it on Windows but did not succeed, I managed to compile it with WSL but could not use it on my project. I tried to use source files (tried to add src and include folders to my project), but got errors with link

dimitrovand, 2025-01-11:
Hello! CCS is primarily an IDE, so the steps can differ based on your project setup. Could you provide more details about your project? For example, what RTOS are you using? Which compiler are you working with (e.g., GCC, TI, etc.)? Are you using CMake or another build system? This information will help in guiding you to integrate libcsp into your project.

Aseke314, 2025-01-13:
Thanks for reply. I use TI compiler V20.2.7.LTS. As the RTOS using FreeRTOS. The microcontroller I use is TMS470MF06607.
I think I need a hekp/hint in two ways:
1. Can someone describe the steps how to use source files of libcsp? 
2. Is someone succeed to compile libcsp in Windows? If yes please also describe the steps

dimitrovand, 2025-01-13:
Thanks for the details! What about your build system? Are you using one of the build systems libcsp is using or a CCS-managed build, and what version of libcsp are you working with?
For your first question: If you're compiling libcsp, CMake, Meson, and Waf already have build scripts available. If you're using CCS-managed build, you'll need to manually add the necessary source and include files to your project + create your own autoconfig.h . To understand how to use libcsp in your application, check out the documentation and provided examples.
For your second question: Using Cygwin might be the easiest approach for building libcsp on Windows. However, keep in mind that some libcsp source code modifications might be required. The latest version of libcsp does not support Windows builds.

yashi, 2025-03-05:
Hi guys. Please reopen this issue if necessary; closing due to inactivity.