libcsp/libcsp #781: How to Preserve the Original Source Address in libcsp 2.0 When Routing Packets Through Intermediate Nodes?

shurik3313, 2025-01-16:
Context:
In libcsp 2.0, when routing packets, the address of the node routing the packet is inserted into the source address field. Consider a network of three devices connected in series: A <-> B <-> C.

Problem:
The goal is to send a message from node A to node C and receive a response from node C. However, the following issue arises during message routing:

Node A sends a message to node C.
When the message passes through node B, the source address in the packet is changed to node B's address.
Node C receives the message and sends a response, but the response is directed to node B (the new source address) instead of node A.
As a result, the message exchange stops at node B, and the response never reaches node A.

Question:
How can this message exchange be properly implemented in libcsp 2.0 to ensure:
The original source address (node A) is preserved throughout the routing process.
The response from node C is delivered directly to node A, bypassing the source address modification at intermediate nodes?

Additional Details:
Are there any built-in mechanisms in libcsp 2.0 to preserve the original source address during routing?
If not, what alternative approaches can be used (e.g., custom fields in the packet, source routing, or manual address management)?
Are there any configuration settings or modifications required in libcsp 2.0 to achieve this behavior?

dimitrovand, 2025-01-24:
Hi @shurik3313, can you please share how you've set up the interfaces on all 3 nodes?

shurik3313, 2025-01-24:
Hi @dimitrovand,

Here’s how I’ve set up the interfaces on all three nodes:

### Node A (STM32, address 30):
Node A is connected to Node B via UART (without KISS). Interface setup code:

```c
csp_iface_t * default_iface_uart = NULL;
int error = uart_driver_init("UART", &default_iface_uart);
if (error != CSP_ERR_NONE) {
    csp_print("failed to add UART interface, error: %d\n", error);
}
default_iface_uart->addr = 30;
csp_rtable_set(0, 0, default_iface_uart, CSP_NO_VIA_ADDRESS);
```
### Node B (STM32, address 40):
Node B is connected to Node A via UART (without KISS) and to Node C via KISS (UART). Interface setup code:
```c
csp_iface_t * default_iface_uart = NULL;
int error = uart_driver_init("MY_UART", &default_iface_uart);
if (error != CSP_ERR_NONE) {
    csp_print("failed to add MY_UART interface, error: %d\n", error);
}
default_iface_uart->addr = 40;

csp_iface_t * iface_kiss = NULL;
csp_usart_conf_t conf = {
    .device = "kiss",
    .baudrate = 115200,
    .databits = 8,
    .stopbits = 1,
    .paritysetting = 0,
    .checkparity = 0
};
error = csp_usart_open_and_add_kiss_interface(&conf, CSP_IF_KISS_DEFAULT_NAME, &iface_kiss);
if (error != CSP_ERR_NONE) {
    csp_print("failed to add KISS interface, error: %d\n", error);
}
iface_kiss->addr = 40;

csp_rtable_set(30, 0, default_iface_uart, CSP_NO_VIA_ADDRESS);  // Route to Node A
csp_rtable_set(50, 0, iface_kiss, CSP_NO_VIA_ADDRESS);         // Route to Node C
```
### Node C (Linux, address 50):
Node C is connected to Node B via KISS (UART). Interface setup code:
```c
csp_iface_t * default_iface = NULL;
csp_usart_conf_t conf = {
    .device = "/dev/ttyAMA0",
    .baudrate = 115200,
    .databits = 8,
    .stopbits = 1,
    .paritysetting = 0
};
int error = csp_usart_open_and_add_kiss_interface(&conf, CSP_IF_KISS_DEFAULT_NAME, 20, &default_iface);
if (error != CSP_ERR_NONE) {
    csp_print("failed to add KISS interface, error: %d\n", error);
    exit(1);
}

default_iface->addr = 50; 
csp_rtable_set(0, 0, default_iface, CSP_NO_VIA_ADDRESS);
```
### Additional Details:
- Node A (address 30) and Node B (address 40) are connected via UART without KISS.
- Node B (address 40) and Node C (address 50) are connected via KISS (UART).
- The routing table on Node B is configured as follows:
  - Route to Node A (address 30) via the UART interface.
  - Route to Node C (address 50) via the KISS interface.

dimitrovand, 2025-01-25:
Looks like an issue with the way your interfaces are set up. Check out the following examples (you need to run them on Linux machine):

nodeb.c:
```
#include <string.h>
#include <unistd.h>
#include <stdlib.h>
#include <pthread.h>
#include <csp/csp.h>
#include <csp/csp_id.h>
#include <csp/csp_debug.h>
#include <csp/drivers/usart.h>

static void * task_router(void * param) {
    (void)param;
    while (1) {
        csp_route_work();
    }
    return NULL;
}

int main(int argc, char * argv[]) {
    uint8_t my_address = 0;
    uint8_t node_a_address = 0;
    uint8_t node_c_address = 0;
    const char * b_to_a_device = NULL;
    const char * b_to_c_device = NULL;
    int opt;
    while ((opt = getopt(argc, argv, "a:b:c:d:e:")) != -1) {
        switch (opt) {
            case 'a':
                node_a_address = atoi(optarg);
                break;
            case 'b':
                my_address = atoi(optarg);
                break;
            case 'c':
                node_c_address = atoi(optarg);
                break;
            case 'd':
                b_to_a_device = optarg;
                break;
            case 'e':
                b_to_c_device = optarg;
                break;
            default:
                exit(1);
        }
    }

    csp_init();

    csp_usart_conf_t b_to_a_uart_conf = {
        .baudrate = 115200,
        .databits = 8,
        .stopbits = 1,
        .paritysetting = 0,
        .device = b_to_a_device
    };

    csp_usart_conf_t b_to_c_uart_conf = {
        .baudrate = 115200,
        .databits = 8,
        .stopbits = 1,
        .paritysetting = 0,
        .device = b_to_c_device
    };

    csp_iface_t * b_to_a_iface = NULL;
    csp_iface_t * b_to_c_iface = NULL;
[... 197 lines cut ...]
                packet->length = 32;
                csp_send(conn, packet);
                
                packet = csp_read(conn, 1000);
                if (packet) {
                    csp_print("received response from node address: %d\n", packet->id.src);
                    csp_buffer_free(packet);
                }
            }

            (void)csp_close(conn);
        }
    }

    return 0;
}
```

You are also have run the following commands in separate terminals:
`socat -d -d -d pty,raw,echo=0,link=/tmp/a_to_b pty,raw,echo=0,link=/tmp/b_to_a`
`socat -d -d -d pty,raw,echo=0,link=/tmp/b_to_c pty,raw,echo=0,link=/tmp/c_to_b`
`nodeb -a 30 -b 40 -c 50 -d /tmp/b_to_a -e /tmp/b_to_c`
`nodec -a 30 -c 50 -d /tmp/c_to_b`
`nodea -a 30 -c 50 -p 10 -d /tmp/a_to_b`
After running these commands you should be able to see, in nodec and nodea terminals, the source address of the received packets.