libcsp/libcsp #782: csp_iflist_get() only finds interface LOOP

lmartinezan, 2025-01-16:
## Description

Hello everyone!
I'm trying to communicate two systems trough Ethernet by using CSP.
Both are connected by RJ45, with IPs in the same range so they can even ping one another. And if I do ifconfig I can see eth0 interface in both.
I have done a basic script just to connect them and get a successful message, but it only finds "LOOP" interface when it searches the interfaces!

## Code

These are the main functions of the script: 
```
void list_interfaces(void) {
    csp_iface_t *ifc = csp_iflist_get(); // Obtain the first interface
    while (ifc) {
        printf("Interface found: %s\n", ifc->name);  // Print the name
        ifc = ifc->next;  // Move to the next interface
    }
}

/* Configuration of eth */
void eth_interface_init(void) {
    list_interfaces();  // Show all the registered interfaces
    csp_iface_t *eth_iface = csp_iflist_get_by_name("eth0"); // Connect to "eth0"
    if (eth_iface == NULL) {
        printf("Error: Not possible to get correct interface\n");
        exit(EXIT_FAILURE);
    }

    printf("Interface correctly  configurated.\n");
}
```
## Expected results - doubt

So, I expect to have the same interface list as with "ifconfig: eth0, lo and wlan0" or at least "eth0 and LOOP" but only "LOOP" is found... and I am not able to start communication among them.

Is it possible to get CSP working over Ethernet interface? or maybe it can only work within the internal interface LOOP?

Thank you! 😊

moonlight83340, 2025-01-17:
Hi,

I’m not sure what the best answer is, but I can explain why you only see the "LOOP" interface.

If you check the libcsp code for `csp_iflist_get()` :
https://github.com/libcsp/libcsp/blob/fc3de68fef5642a2e195351a306e2e2dfe3f5f17/src/csp_iflist.c#L203
libcsp returns all interfaces that have been registered.

So why do you only have "LOOP"?
The simple answer is that "LOOP" is added during the initialization of CSP:
https://github.com/libcsp/libcsp/blob/fc3de68fef5642a2e195351a306e2e2dfe3f5f17/src/csp_init.c#L45

If you want to add an interface, you’ll need to do it yourself:
https://github.com/libcsp/libcsp/blob/fc3de68fef5642a2e195351a306e2e2dfe3f5f17/src/csp_iflist.c#L155

That said, you can also check the currently supported interfaces (this is handled in a helper function):
https://github.com/libcsp/libcsp/blob/fc3de68fef5642a2e195351a306e2e2dfe3f5f17/src/drivers/eth/eth_linux.c#L178

The helper function : `csp_eth_init` is probably what you want to use.
https://github.com/libcsp/libcsp/blob/fc3de68fef5642a2e195351a306e2e2dfe3f5f17/src/drivers/eth/eth_linux.c#L70

I hope this provides a satisfying answer.

moonlight83340, 2025-01-30:
@lmartinezan did you resolve your problem ?
I have done a PR for ethernet interface example, https://github.com/libcsp/libcsp/pull/783
It could help you how to use the interface, in the github workflow I use a veth interface for testing purpose.