libcsp/libcsp #794: libcsp 2.0 CAN FD support

IlievIliya92, 2025-02-09:
Hello everyone, 

I was wondering if libcsp 2.0. supports CAN FD and in the case it doesn't what modifications would be required to use this protocol under Linux and FreeRTOS?

My current understanding is that CAN 2.0 is supported only and in order to use the FD extension both the `interfaces/csp_if_can.c` and the `drivers/can_socketcan.c` modules would require some extra functionality to be added?

moonlight83340, 2025-02-10:
You're right, currently libcsp support only CAN 2.0, 
https://github.com/libcsp/libcsp/blob/develop/src/drivers/can/can_socketcan.c#L61
It would need some modification to add the CAN FD supports. (Not sure yet how much)

moonlight83340, 2025-04-14:
@IlievIliya92
I attempted to add CAN FD support to libcsp, focusing initially on the Unix implementation. So far, I've tested it using virtual CAN interfaces. 
You can find the pull request here: https://github.com/libcsp/libcsp/pull/835.
I hope this contribution is helpful; there might still be some modifications needed.
I plan to test it on physical CAN FD hardware once I have access to it.​

IlievIliya92, 2025-04-17:
@moonlight83340 Thank you very much for the information, the effort and the valuable pull request!

closed 2025-04-17 by IlievIliya92