libcsp/libcsp #802: Zephyr server-client over CAN example with STM32 Nucleo Boards

AbdellahEssetty, 2025-02-27:
## Description

**Hardware:**
- STM32 Nucleo-G431RB
- STM32 Nucleo-H7A3ZI-Q
- TJA1042 CAN Tranceiver

**Description:**
I am attempting to run the libcsp server-client sample on Zephyr (v4.0.0) using two STM32 Nucleo boards (Nucleo-G431RB and Nucleo-H7A3ZI-Q). While the physical CAN bus is functioning and transmitting data correctly (confirmed via oscilloscope output), I am encountering issues with the sample's operation.

I would like to know if there are any modifications required to make the libcsp server-client sample work correctly on Zephyr 4.0.0 with these boards.

## Steps to Reproduce

1. Removed the reboot command sent from the client to the server.
2. Device Tree Overlay (Used for Both Server and Client):
```dts
/ {
    chosen {
        zephyr,canbus = &fdcan1;
    };

    can_phy2: can-phy2 {
        compatible = "can-transceiver-gpio";
        enable-gpios = <&gpiob 8 GPIO_ACTIVE_LOW>;
        standby-gpios = <&gpiob 9 GPIO_ACTIVE_HIGH>;
        max-bitrate = <8000000>;
        #phy-cells = <0>;
    };
};

&fdcan1 {
    pinctrl-0 = <&fdcan1_rx_pb8 &fdcan1_tx_pb9>;
    pinctrl-names = "default";
    bus-speed = <1000000>;
    status = "okay";
    phys = <&can_phy2>;
};
```
3. Server configuration (Nucleo G431RB):
    * address = 2
    * server_address = 255
    * filter_addr = 0x3FFF
    * filter_mask = 0x0000
4. Client configuration (Nucleo H7A3ZI_Q):
     * address = 1
    * server_address = 2
    * filter_addr = 0x3FFF
    * filter_mask = 0x0000

## Expected behavior

I expected the server to receive the message and to respond

## Actual Behavior

There were no respond from the server, and the client kept sending messages. 

## Environment Details

Provide information about your environment, including:

- Build OS:  Debian GNU/Linux 12 (bookworm)
- Target OS: Zephyr
- libcsp commit hash: 63e1b05
- Meson version: None
- CMake version: 3.25.1
- GNU Make version: 4.3
- Ninja version: 1.11.1
- DTC version: 1.6.1
- Zephyr version:  4.0.0-rc1
- West version: 1.2.0

## Logs or console output from the very beginning to the end

Client-side log:
```

[00 0x240046f0] S:0, 32827 -> 60714, 220 -> 148 (17) fl 0
[01 0x240047c4] S:0, 2 -> 0, 18 -> 1 (18) fl 0
[02 0x24004898] S:0, 59436 -> 5547, 125 -> 249 (19) fl 0 
[03 0x2400496c] S:0, 54871 -> 41410, 239 -> 133 (20) fl 0 
[04 0x24004a40] S:0, 36980 -> 2097, 87 -> 38 (21) fl 0 
[05 0x24004b14] S:0, 54148 -> 59385, 218 -> 109 (22) fl 0 
[06 0x24004be8] S:0, 24626 -> 5103, 15 -> 230 (23) fl 0 
[07 0x24004cbc] 5:0, 62330 -> 55104, 52 -> 150 (24) fl 0
LOOP		addr: 0 netmask: 14 dfl: 0
		tx: 00000 IX: 00000 txe: 00000 rxe: 00000 
		drop: 00000 autherr: 00000 frame: 00000 
		txb: 0 (0B) Ixb: 0 (0B)

CAN0		addr: 1 netmask: 0 dfl: 1
		tx: 00000 IX: 00000 txe: 00000 Ixe: 00000 
		drop: 00000 autherr: 00000 frame: 00000 
		txb: 0 (0B) Ixb: 0 (0B)

Booting Zephyr OS build v4.0.0-rc1-10-gafce5329f772 *** 
[00:00:00.000,000] <inf> csp_sample_server_client: Initialising CSP 
[00:00:00.000,000] <inf> csp_sample_server_client: Connection table 
[00:00:00.040,000] <inf> csp_sample_server_client: Interfaces
[00:00:00.072,000] <inf> csp_sample_server_client: Client task started
```

Server-side log:
```

[00 0x200046b0] S:0, 61440 -> 63585, 43 -> 70 (17) fl 0 
[01 0x20004784] S:0, 61699 -> 513, 3 -> 68 (18) fl 0 
[02 0x20004858] S:0, 17050 -> 18048, 6 -> 209 (19) fl 0
[03 0x2000492c] S:0, 61440 -> 20728, 151 -> 178 (20) f1 0 
[04 0x20004a00] S:0, 26795 -> 25155, 168 -> 96 (21) fl 0 
[05 0x20004ad4] S:0, 33553 -> 62340, 18 -> 136 (22) fl 0 
[06 0x20004ba8] S:0, 25003 -> 17977, 48 -> 70 (23) fl 0
[07 0x20004c7c] S:0, 16645 -> 18005, 234 -> 145 (24) fl 0 
LOOP addr: 	addr: 0 netmask: 14 dfl: 0
		tx: 00000 rx: 00000 txe: 00000 rxe: 00000 
		drop: 00000 autherr: 00000 frame: 00000 
		txb: 0 (0B) rxb: 0 (0B)

CAN0		addr: 2 netmask: 0 dfl: 1
		tx: 00000 rx: 00000 txe: 00000 rxe: 00000 
		drop: 00000 autherr: 00000 frame: 00000 
		txb: 0 (0B) rxb: 0 (0B)

***Booting Zephyr OS build v4.0.0-rc1-10-gafce5329f772 *** 
[00:00:00.000,000] <inf> csp_sample_server_client: Initialising CSP 
[00:00:00.000,000] <inf> csp_sample_server_client: Connection table 
[00:00:00.040,000] <inf> csp_sample_server_client: Interfaces
[00:00:00.073,000] <inf> csp_sample_server_client: Server task started
```

## Additional Context
- The physical CAN bus appears to be functioning correctly as verified through an oscilloscope capture. The header and the content of the frames seem right, but there is no acknowledgment.

[image]

## Checklist

- [x] I've searched in "Issues" for similar issues, both open and closed.
- [x] I've used GitHub Flavored Markdown (GFM) to format my text.
- [x] I've avoided using screenshots for console output or logs
- [x] I've included all relevant details and steps to reproduce the issue.
- [x] I've provided information about my environment.
- [x] I've attached necessary logs or code snippet for minimal reproducible example.

yashi, 2025-03-05:
let me reproduce it.

IlyaYakushevskiy, 2025-03-22:
+1 , 
I try to reproduce it with two STM32 boards (nucleo_h723zg) on Zephyr. The issue is caused csp_accept() function, it just gets stuck on trying updating the queue. I've never managed for server to get a connection on any socket in any circumstances. 
Albeit using standard zephyr/drivers/can.h I have absolutely no problem exchanging data between two boards over CAN.

moonlight83340, 2025-03-25:
>     1. Removed the reboot command sent from the client to the server.
DId you also removed `csp_ping ` ?

> ## Expected behavior
> 
> I expected the server to receive the message and to respond
> ## Actual Behavior
> 
> There were no respond from the server, and the client kept sending messages.
> ## Environment Details

In the current sample, the server will not answer the client but print with `LOG_INF`.

Could you share your diff ? or your branch with your modifications ?

Thank you,

AbdellahEssetty, 2025-03-25:
> 
> DId you also removed `csp_ping ` ?
> 
No, I have only removed the reboot.
> 
> In the current sample, the server will not answer the client but print with `LOG_INF`.
> 
Shouldn't it at least acknowledge the CAN frames? Anyway, the console of the server doesn't log anything. 
>
> Could you share your diff ? or your branch with your modifications ?
> 
Client Diff

```Diff
...
-  csp_reboot(server_address);
-  LOG_INF("reboot system request sent to address: %u", server_address);
...
-  address = 10;
-  server_address = 255;
-  const struct device * device = DEVICE_DT_GET(DT_NODELABEL(can0));

+  address = 1;
+  server_address = 2;
+  const struct device * device = DEVICE_DT_GET(DT_NODELABEL(fdcan1));
...
uint16_t filter_addr = address;
uint16_t filter_mask = 0x3FFF;
+  filter_addr = 0x3FFF;
+  filter_mask = 0x0000;
```

Server Diff
``` Diff
...
-  csp_reboot(server_address);
-  LOG_INF("reboot system request sent to address: %u", server_address);
...
-  address = 10;
+  address = 2;
server_address = 255;
-  const struct device * device = DEVICE_DT_GET(DT_NODELABEL(can0));
+  const struct device * device = DEVICE_DT_GET(DT_NODELABEL(fdcan1));
...
uint16_t filter_addr = address;
uint16_t filter_mask = 0x3FFF;
+  filter_addr = 0x3FFF;
+  filter_mask = 0x0000;
```

> 
> Do you have the necessary termination resistors (120Ω) in place? Normally, a CAN bus requires two 120Ω resistors, one at each end of the bus. If you're directly connecting two STM32 devices, do you have one resistor on each device (a total of two)?
> 
I have used a module with built-in 120Ω resistors. As I have mentioned, the client does send successfully the frames. So it seems to me that the physical layer works properly. I have added 10KΩ resistors on tx and rx (for protection purposes). 

[image]

Thank you,

moonlight83340, 2025-03-25:
@AbdellahEssetty 
I just notice something, I think your `filter_addr `and `filter_mask `. Could be the problem. Did you try it with the example one ?

moonlight83340, 2025-03-26:
> No, I have only removed the reboot.

Do you have some more logs on client side ? you should have ping failing if it can't reach the server.

> Shouldn't it at least acknowledge the CAN frames? Anyway, the console of the server doesn't log anything.

you should indeed have some log on the packet received.

Thanks, 

I have try with an STM32 F429zi  server-client example against a PC server/client with and usb can, but I don't have the same issue as you.
I will try to test with two boards.

AbdellahEssetty, 2025-03-26:
> 
> Do you have some more logs on client side ? you should have ping failing if it can't reach the server.
> 
Not really, I have modified the program further, and printed the connection table and the interfaces information continuously in the while loop, but nothing changed. Though on very few occasions, the client output the error message
```
<err> csp_can_driver: [CAN0] can_send() failed, errno -114
```
Using zephyr error code, it showed that it represents an *Unreachable Network* error. So I suppose it was just some malfunction on the hardware side (maybe a jumper wire). But everything works well after adjusting the connections over the breadboard.

I suspect the problem is using an ISR function to enqueue the message, or it could be in the device tree overlay file. But I still need further investigation. 

Or less likely there may be a problem with stm32h7 in Zephyr, I will try to use another stm32g4 board and check. 

Thanks.

moonlight83340, 2025-03-26:
> Not really, I have modified the program further, and printed the connection table and the interfaces information continuously in the while loop, but nothing changed. Though on very few occasions, the client output the error message

Not sure if it will help but could you share here your client log, with error ?

Thank you for many informations, I will take a look again.

moonlight83340, 2025-03-26:
if it's always `result -1` , it definitively can't reach the server.

Thank you for your time also,

AbdellahEssetty, 2025-03-26:
Yes, it has always been -1.

AbdellahEssetty, 2025-03-27:
> 
> Could you add a LOG_INF here? https://github.com/libcsp/libcsp/blob/develop/src/drivers/can/can_zephyr.c#L62
> 
> Something like this: `LOG_INF("[%s] k_msgq_get() packet\n", iface->name);`
> 
I added the line, and the log got printed multiple times.

Server-side log
```
[00 0x240046f0] S:0, 40991 -> 6116, 85 -> 243 (17) fl 0
[01 0x240047c4] S:0, 62762 -> 60775, 232 -> 254 (18) fl 0
[02 0x24004898] S:0, 56969 -> 49841, 173 -> 193 (19) fl 0
[03 0x2400496c] S:0, 13 -> 16215, 102 -> 245 (20) fl 0
[04 0x24004a40] S:0, 54423 -> 16303, 127 -> 168 (21) fl 0
[05 0x24004b14] S:0, 27550 -> 13285, 197 -> 158 (22) fl 0
[06 0x24004be8] S:0, 36633 -> 16590, 148 -> 183 (23) fl 0
[07 0x24004cbc] S:0, 2357 -> 58378, 24 -> 217 (24) fl 0
LOOP       addr: 0 netmask: 14 dfl: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B)

CAN0       addr: 2 netmask: 0 dfl: 1
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B)

*** Booting Zephyr OS build v4.0.0-rc1-10-gafce5329f772 ***
[00:00:00.000,000] <inf> csp_sample_server_client: Initialising CSP
[00:00:00.000,000] <inf> csp_sample_server_client: Connection table
[00:00:00.041,000] <inf> csp_sample_server_client: Interfaces
[00:00:00.073,000] <inf> csp_sample_server_client: Server task started
[00:00:00.757,000] <inf> csp_can_driver: [CAN0] k_msgq_get() packet

[00:00:00.757,000] <inf> csp_can_driver: [CAN0] k_msgq_get() packet

[00:00:00.757,000] <inf> csp_can_driver: [CAN0] k_msgq_get() packet

[00:00:00.757,000] <inf> csp_can_driver: [CAN0] k_msgq_get() packet

[00:00:00.757,000] <inf> csp_can_driver: [CAN0] k_msgq_get() packet

[00:00:00.758,000] <inf> csp_can_driver: [CAN0] k_msgq_get() packet
```

Client-side log
```
[00 0x200046b0] S:0, 2 -> 0, 17 -> 10 (17) fl 0
[01 0x20004784] S:0, 2 -> 0, 18 -> 1 (18) fl 0
[02 0x20004858] S:0, 2 -> 0, 19 -> 10 (19) fl 0
[03 0x2000492c] S:0, 2 -> 0, 20 -> 1 (20) fl 0
[04 0x20004a00] S:0, 2 -> 0, 21 -> 10 (21) fl 0
[05 0x20004ad4] S:0, 2 -> 0, 22 -> 1 (22) fl 0
[06 0x20004ba8] S:0, 2 -> 0, 23 -> 10 (23) fl 0
[07 0x20004c7c] S:0, 2 -> 0, 24 -> 1 (24) fl 0
LOOP       addr: 0 netmask: 14 dfl: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B)

CAN0       addr: 1 netmask: 0 dfl: 1
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B)

*** Booting Zephyr OS build v4.0.0-rc1-10-gafce5329f772 ***
[00:00:00.000,000] <inf> csp_sample_server_client: Initialising CSP
[00:00:00.000,000] <inf> csp_sample_server_client: Connection table
[00:00:00.034,000] <inf> csp_sample_server_client: Interfaces
[00:00:00.067,000] <inf> csp_sample_server_client: Client task started
[00:00:02.067,000] <inf> csp_sample_server_client: -
[00:00:02.072,000] <err> csp_can_driver: [CAN0] can_send() failed, errno -114
[00:00:03.072,000] <inf> csp_sample_server_client: Ping address: 2, result -1 [mS]
[00:00:03.077,000] <err> csp_can_driver: [CAN0] can_send() failed, errno -114
```

While testing, I observed an interesting behavior: after a couple of minutes, the oscilloscope indicated that one single CAN frame was acknowledged. However, neither the client nor the server logged any related event.

[image]

moonlight83340, 2025-03-27:
ok thank you,
So based on your log, your're getting the can frame.
So we will have to dig a bit on the can interface I think, to verify if they are accepted as csp packet.
Can you add some log here : 
https://github.com/libcsp/libcsp/blob/6b6e2dd2870bd1f8ab4516123f1fa255d6c1b632/src/interfaces/csp_if_can.c#L261
I mean every time there is an escape route with an error to watch if you're packet aren't accepted.

`csp_can_driver: [CAN0] can_send() failed, errno -114`
weird it's an error when network isn't reach... but most of the time you seems to reach the server but you don't keep the packet..

AbdellahEssetty, 2025-04-07:
Sorry for the late response, I couldn't do much last week. 
However, I don't know what happened, but everything started working suddenly after a couple of resets. Nothing has changed in the software (except sudo apt update & upgrade).

It may have been a hardware issue after all (or extremely less likely an software issue with debian).

Shall I close the issue?

moonlight83340, 2025-04-07:
Thank you for letting us know,
If everything is fine, you can close the issue. But if you run into any trouble again, don't hesitate to open a new one — or reopen this one if it's the same problem.

moonlight83340, 2025-04-08:
> +1 , I try to reproduce it with two STM32 boards (nucleo_h723zg) on Zephyr. The issue is caused csp_accept() function, it just gets stuck on trying updating the queue. I've never managed for server [...]

This issue appears to be resolved for the original reporter.

If you're still encountering a problem, please open a new issue and include as much information as possible about your setup, logs, and how to reproduce the issue. That way, we can help you debug more effectively.

Thanks!